Opentrons/opentrons #4811: Modules cannot be used in API V2 protocols run on Jupyter

ethanfjones, 2020-01-21:
## overview

When you try to run an API V2 protocol that contains a module(in this case the mag deck) the below error is thrown:

RuntimeError: Could not find specified module: magdeck

## steps to reproduce

Verify module is connected to the robot in the app
Open up your robot's jupyter notebook server 
Open a new notebook
Try to run the following protocol 

```python
from opentrons import protocol_api
import opentrons.execute

metadata = {'apiLevel': '2.0'}

def run(protocol: protocol_api.ProtocolContext):
    mag_mod = protocol.load_module('magnetic module', '1')
    plate = mag_mod.load_labware('nest_96_wellplate_100ul_pcr_full_skirt')

    
    
    mag_mod.engage()
    
    
protocol = opentrons.execute.get_protocol_api('2.0')
run(protocol)  
```

## expected behavior

You should be able to run API V2 protocols that contain a module in your robot's Jupyter notebook.

mcous, 2021-07-27:
Note to team: Is this at all related to #8009?

mcous, 2021-12-10:
Closed as resolved

closed 2021-12-10 by mcous