Opentrons/opentrons #4837: Clean up flow rate terminology and semantics

SyntaxColoring, 2020-01-23:
# Overview

The flow rate adjustment situation is kind of messy.

## "Flow rate" vs. "Speed"

The API [has a distinction](https://docs.opentrons.com/v2/new_pipette.html?highlight=pipette%20flow_rate#plunger-flow-rates) between "speed" and "flow rate."

* `pipette.speed.aspirate` et. al. adjust the plunger speed in mm/sec, and is probably *not* what people want to use.
* `pipette.flow_rate.aspirate` et. al. adjust the plunger speed in µL/sec.

These terms are too easily conflated.
* In internal communications, we use them interchangeably.
* In the API documentation, we use them sloppily and wrongly.
* In the run log, we wrongly say "speed" instead of "rate."

## Absolute value vs. relative multiplier

* `pipette.flow_rate.aspirate = 10` sets the pipette's aspirate flow rate to 10 µL/sec.
* `pipette.aspirate(vol, well, rate=10)` aspirates 10 times faster than the default, whatever the default is.

This discrepancy is confusing. Plus, it's not always an easy mistake to notice when you watch the robot run through a protocol.

Also, the run log says "Aspirating...**at speed 1.0**". First, as described above, "speed" should be "rate." But also, the 1.0 is apparently just the raw `rate=` argument. It doesn't take into account any changes made to `pipette.flow_rate.aspirate`.

# Proposals

1. For specificity and correctness, rename the liquid handling functions' `rate` kwarg to something like `flow_rate_multiplier`, and deprecate the old name.
2. For completeness, add a `ul_per_sec` kwarg to the liquid handling functions.
3. For consistency with <span>#</span>2, rename `pipette.flow_rate.aspirate` et. al. to `pipette.ul_per_sec.aspirate`, and deprecate the old name.
4. Change the run log to say "Aspirating...at 123 µL / sec." (I'm assuming that the absolute number is what people are most likely to care about. If we still want to mention the multiplier, we should think of a better way to say it, like "Aspirating...at 200% the default rate" or something.)

## Outside-the-box proposal

Have fewer functions with kwargs and use context managers instead.

```python
pipette.dispense(loc, 100)
with pipette.ul_per_sec(aspirate=50, dispense=25):
    pipette.mix(repetitions=4)
```

# These terms in other contexts

These are off the top of my head. We should probably do some greps to be exhaustive.

## In the API documentation

Examples and documentation use the terms "speed" and "rate" interchangeably. They should mostly be unified to say "flow rate." We can say "speed" in a few calculated places for searchability.

## In support articles

There's probably some support articles that make the same mistake.

## In Protocol Designer

Protocol Designer seems inconsistent, using "Flow Rate" as a label on the main step screen and "Aspirate Speed" in a popup. They should all probably be changed to "Flow Rate." We can ticket that separately once we're clear on what to do here.

IanLondon, 2020-01-24:
Thanks Max!

Related: https://github.com/Opentrons/opentrons/blob/d07efcba8522ee7e0178a68342dc81fb4998387e/api/src/opentrons/protocol_api/execute_v3.py#L33

Since there's no nice way to specify flow rate in uL/sec in blowout/aspirate/dispense, the JSON executor sets the flow rate before blowout/aspirate/dispense in a boilerplate-heavy way. Would be great to have a kwarg to set absolute uL/sec instead of relative %!

SyntaxColoring, 2020-05-02:
A small part of this was previously ticketed as #2772. Updated this ticket accordingly.

MarcelRobitaille, 2021-02-18:
I agree. Another thing is that the docs for `aspirate` say that the plunger speed will be `rate * aspirate_speed`, but there is no other mention of `aspirate_speed` in the docs. How do I set it?

SyntaxColoring, 2021-02-24:
@MarcelRobitaille, I think that's a mistake in the docs. Instead of `aspirate_speed`, it should say [`speed.aspirate`](https://docs.opentrons.com/v2/new_protocol_api.html#opentrons.protocol_api.contexts.InstrumentContext.speed).

Thanks for asking about that. We'll get it fixed soon.

SyntaxColoring, 2021-03-29:
In our overhaul of the Protocol API back-end ("Protocol Engine"), we're trying to use µL/sec flow rates instead of multipliers. This should address internal DX like what @IanLondon brought up, but won't directly address the DX of the user-facing Python Protocol API.

closed 2022-09-30 by bndo