Opentrons/opentrons #4838: Expose the pipette serial number through the API

SyntaxColoring, 2020-01-23:
# Overview

Python API protocols should be able to programmatically access the pipette's serial number, like "P10MV1519042304", through a property like `pipette.serial_number`.

This would expand the information already exposed by `pipette.name` and `pipette.model`.

# Why

This request comes from a customer!

They have three OT-2s: one for human research, one for secret dinosaur research, and one for secret Martian research. Each OT-2 has its own designated set of pipettes. To mitigate cross-contamination, they want to prevent protocols written for one robot from accidentally using pipettes meant for other robots.

Protocols would be written for specific pipettes. If the wrong pipette is attached, the protocol would raise an error.

mcous, 2020-01-23:
I think this might already be available via the `hw_pipette` property of PipetteContext

```
pipette = protocol.load_instrument('p300_single_gen2', 'right')
pipette_id = pipette.hw_pipette.get('pipette_id')
```

SyntaxColoring, 2020-01-24:
@mcous That looks like a good immediate solution.

Do you intend that snippet to be the Opentrons-supported way to do this, or is it just a workaround/hack?

mcous, 2020-01-24:
I would call this a stable-ish workaround. We're not going to break the interface, but having a top-level "official" method to the PipetteContext to expose serial number needs more thought before we can figure out something we consider "good" and "safe"

Also I would keep an eye out for the returned value during simulation. I think it depends on having the correct pipettes attached at simulation time, so something like CLI calibration I think would return a dummy value

SyntaxColoring, 2020-02-25:
The analogous workaround in APIv1 is:
```python
from opentrons import robot
if not robot.is_simulating():
    pipettes = robot.get_attached_pipettes()
    serial_number = pipettes['left']['id']
```

closed 2022-09-30 by bndo