Opentrons/opentrons #4873: LC: Update PDF to new test protocol instructions

howisthisnamenottakenyet, 2020-01-28:
Acceptance criteria:
- [ ] Design to updated PDF
- [ ] Button links to this guide in LC need to be updated to this new pdf

howisthisnamenottakenyet, 2020-01-28:
@pantslakz @umbhau Could one of you take a quick peek at the PDF? It's a long doc so I'd love a second set of eyes to see help check if I missed anything while updating it.

pantslakz, 2020-01-29:
Reads great @howisthisnamenottakenyet 

If I understand correctly - the main difference now is users can set up deck from start - right?

I do wonder how we can effectively communicate this change to prior LC users, so they know to access the new instructions/not be surprised by movement changes.

howisthisnamenottakenyet, 2020-01-29:
@pantslakz Yup! I'm hoping the visual change on the deck from slot 3 to slot 2 will help cue them in

Kadee80, 2020-02-12:
Closed via AWS update.

closed 2020-02-12 by Kadee80