Opentrons/opentrons #4922: App and robot version labels are confusing

mcous, 2020-02-05:
## overview

As a user, I would like to understand what the different software versions reported by the app mean.

As reported by @SyntaxColoring:

* The Opentrons App currently labels its own version as "Software Version." This is ambiguous, and has been conflated with the *robot's* software version (despite being under More > App)
* Users have seen "Firmware version" and assumed that that's the "version of the robot software." In fact, it's specifically the firmware on the motor controller board, which is only rarely of interest

This issue closes #4806 and takes over for that proof-of-concept PR.

## acceptance criteria

- [ ] App version is reported to the user and the fact that it's the version of the software running on the user's computer is unambiguous
- [ ] Robot software version is reported to the user and the fact that it's the version of the software running on the robot is unambiguous

## implementation details

- This existing "firmware version" UI was introduced at a time when API updates were applied separately from Smoothie firmware updates
    - Currently, Smoothie firmware is applied automatically by the server, and there should be a strict 1:1 relationship between server version and Smoothie version
    - This may indicate that "firmware version" demoted to a tooltip or removed entirely
- SWE would like to get to a place this year where app releases can have a separate release candence from robot software releases, meaning app + robot versions will no longer match
    - This increases the importance of having unambiguous version reporting

SyntaxColoring, 2020-02-06:
One more implementation detail.

The current term *API server version* is a misnomer: though it sounds like it only applies to the API server, it actually applies to the whole system running on the OT-2.

However, the alternatives we've come up with so far aren't great.
* *Robot software* version is too easily conflated with the Opentrons App.
* *Robot operating system* version is perhaps too technical, and is also confusable with [another implementation](https://www.ros.org/).
* *Firmware* version is perhaps technically incorrect, since it's a full Linux system, not some embedded firmware.

-----

That said, I don't think all of these problems have to be solved at once. We can improve the labeling incrementally, deferring the tricky stuff until we have a good answer for it.

alexjoel42, 2021-05-21:
Hi! 
I was wondering if this could be addressed through documentation. Currently there are three main issues that I have seen users of the OT-2 have if they don't know to look out for a difference in server version from app version. 

1. GEN2 Multi-Channel pipettes are reported to support as "broken" when their OT-2 is just on 3.16.2. 
2. Difficulty connecting to the OT-2 on Wi-Fi
3. Inability to calibrate

This is a draft of the documentation that I am working on to see if we can address this through documentation as opposed to changing the labeling. 
https://docs.google.com/document/d/1WHmd2dOW1OCc2p0rG0Ec1bvz70A4Fj406G9sid7eYV8/edit?usp=sharing 

Note: This could be referenced in the Get Started Guide as a warning if deck calibration or pipette attachment doesn't work.

SyntaxColoring, 2021-06-02:
@alexjoel42 

We could certainly try mitigating this with documentation, but I'm skeptical that it would help enough to be worthwhile.

Because:

* The confusion is inherently very contextual. Users won't know that they need to consult the documentation; they'll just see the app labels and immediately make the wrong conclusion.
  * We could add a link or a tooltip to the app. But...if we're modifying the app, why couldn't we just fix the labels in the first place? And permalinks have maintainability problems.
* You can see the underlying problem as the app not providing a language foundation that's sufficient for the documentation to build on. Even if we discuss this at length in the documentation, we'll forever be fighting against the underlying terminology not making sense.

Also note that the conceptual difference between updating your app and updating your robot is sort of already introduced in [Get started: Update your OT-2](https://support.opentrons.com/en/articles/1795303-get-started-update-your-ot-2). But it doesn't show you where to find those two version numbers in the app.

mcous, 2022-06-06:
Closing as stale due to UI overhaul. Taking a peek at it, I _think_ the newer UI does a better job of communicating versions

closed 2022-06-06 by mcous