Opentrons/opentrons #4940: blow_out parameter in transfer() defaults to trash

opatel0, 2020-02-07:
## overview
Bug with transfer() command

## current behavior
When blow_out=True is passed in a transfer() command, the default destination of the blow out is in the Fixed Trash

## steps to reproduce
Run or simulate the following code on OT-2 and observe the simulator's output or run the protocol
```
from opentrons import protocol_api

def run(protocol):
    plate = protocol.load_labware('corning_96_wellplate_360ul_flat', 1)
    tiprack = protocol.load_labware('opentrons_96_tiprack_20ul', 2)

    p20s = protocol.load_instrument('p20_single_gen2', mount='left', tip_racks=[tiprack])
    p20m = protocol.load_instrument('p20_multi_gen2', mount='right', tip_racks=[tiprack])

    p20m.transfer(20, plate['A1'], plate['A2'], blow_out=True)
    p20s.transfer(20, plate['A1'], plate['A2'], blow_out=True)
```

## expected behavior
When blow_out=True is passed in transfer(), the pipette should blow out in the destination well of each transfer

SyntaxColoring, 2020-07-27:
For differences between this and #5170, see https://github.com/Opentrons/opentrons/issues/5170#issuecomment-649047676.

asogluizzo, 2020-10-07:
This will be fixed in 4.0.0 release

closed 2020-12-02 by asogluizzo