Opentrons/opentrons #4946: Docs: Explain how multi-channel pipettes work

SyntaxColoring, 2020-02-07:
The API documentation doesn't sufficiently explain how to use multi-channel pipettes.

There are two parts to this:

# 1. For building block commands

The building block commands all take single `Well` objects as locations. For multi-channel pipettes, that single well should be the one that we want the back-most channel to be over. Like:

```python
p300.aspirate(300, plate.rows()[0][n])
```

As far as I can tell, this just isn't documented. So, users need support's help to tell them how to use building block commands. They try to apply the same patterns that they use for complex commands, and then run into errors.

# 2. For complex commands

Users generally *are* able to figure out on their own how to use multi-channels with the complex commands – probably by following the patterns they see in examples. However, I'm bringing them up here because I personally find things too vague for comfort.

Off the top of my head, roughly in order of descending priority:

1. When users have a choice, should users specify a location as a single well, or as a column of wells? 
    * Why would users choose one over the other?
    * Why do we support both?
    * The type annotations don't account for the ability to pass columns: they use `Well` where they should be using `Union[Well, List[Well]]`. Does that imply passing columns is deprecated?
1. How do things that aren't 96 well plates work? Especially 384 well plates (or, in general, things with more than 8 rows).
    * What does it mean to transfer to or from a column of 16? Will a transfer go `source->even_dests->odd_dests` or `source->even_dests, source->odd_dests`?
    * What does it mean to transfer to or from a *list* of columns of 16?
    * How does this interact with things like `new_tip` and `carryover`?
1. Columns are just lists of wells. In general, how does this work with the parts of complex commands that are documented to change their behavior depending on whether they get a list or a single well?
1. What happens when I pass a [list of volumes](https://docs.opentrons.com/v2/new_complex_commands.html#list-of-volumes)?
1. What happens if I try to use a column from something with fewer rows than a 96 well plate?
1. What happens if I try to use a column from labware whose wells don't have 9 mm pitch?
1. What does custom labware need to do to support multi-channel pipettes as well as standard laware?
1. What happens if I pass a truncated column?
1. What happens if I pass a column with some wells missing from the middle?
1. What happens if I pass a column in non-top-to-bottom order?

SyntaxColoring, 2020-03-26:
A specific confusion that I've seen a few times is that people expect `multi.pick_up_tip(rack.columns()[n])` to work. It doesn't, because `pick_up_tip()` is a building block command that expects to be given a single well (the one for the back-most channel).

nickcrider, 2020-04-21:
From a user standpoint, I feel like something like this should be expected to JustWork(tm) work too:
`for i in range(1,5):
        p_multi.pick_up_tip()
        p_multi.aspirate(100, plate_96well.columns_by_name()[i] )
        p_multi.dispense(100, plate_96well.columns_by_name()['10'] )
        p_multi.return_tip()`

nickcrider, 2020-05-01:
The fact that this doesn't work I am guessing is related. x-posting from this ticket: https://github.com/Opentrons/opentrons/issues/5461

p50.transfer(50, 
     reaction_plate.columns_by_name()['1'].bottom(), 
     mag_plate.columns_by_name()['1']
)

PR #6175 (2020-07-21): docs(api): explain multichannel pipettes
Adds a block of explanation about the way users should specify locations when using multichannel pipettes 

Closes #4946
commit c4f3874e (2020-07-21): docs(api): explain multichannel pipettes (#6175)

Add documentation about the peculiarities of interacting with multichannel pipettes in the protocol api.

Closes #4946
closed 2020-07-21 by sfoster1