Opentrons/opentrons #4981: Scopes act weird in protocols, rejecting valid Python

SyntaxColoring, 2020-02-12:
This is a valid Python script. (Try executing it with `python scope_test.py`.)

```python
metadata = {"apiLevel": "2.1"}

deck_clearance = 0.5
calibration_cross_coordinates = [(x, y, deck_clearance) for x, y in [
    (12.13, 9.0),
    (380.87, 9.0),
    (12.13, 348.5)
]]

def run(protocol):
    protocol.comment(str(calibration_cross_coordinates))

print(calibration_cross_coordinates) # Output something as a sanity check when running as a Python script.
```

But, if you try to execute it with `opentrons_simulate scope_test.py`:

```
Traceback (most recent call last):
  File "/Users/maxpm/.pyenv/versions/3.6.9/bin/opentrons_simulate", line 8, in <module>
    sys.exit(main())
  File "/Users/maxpm/.pyenv/versions/3.6.9/lib/python3.6/site-packages/opentrons/simulate.py", line 489, in main
    log_level=args.log_level)
  File "/Users/maxpm/.pyenv/versions/3.6.9/lib/python3.6/site-packages/opentrons/simulate.py", line 316, in simulate
    execute.run_protocol(protocol, context)
  File "/Users/maxpm/.pyenv/versions/3.6.9/lib/python3.6/site-packages/opentrons/protocol_api/execute.py", line 126, in run_protocol
    _run_python(protocol, context)
  File "/Users/maxpm/.pyenv/versions/3.6.9/lib/python3.6/site-packages/opentrons/protocol_api/execute.py", line 92, in _run_python
    exec(proto.contents, new_globs, new_locs)
  File "scope_test.py", line 7, in <module>
    (12.13, 348.5)
  File "scope_test.py", line 4, in <listcomp>
    calibration_cross_coordinates = [(x, y, deck_clearance) for x, y in [
NameError: name 'deck_clearance' is not defined
```

Maxpyt, 2020-02-25:
This behavior is also observed when trying to evaluate a function inside a class definition.

amitlissack, 2020-02-27:
Worth noting that this isn't something that we're doing in the simulator. I made this dumb fixture:

```
from sys import argv

def runit(fname):
	with open(fname) as f:
		p = f.read()
		exec(p)

if __name__ == "__main__":
	runit(argv[1])

```
running `python fixture.py scope_test.py` yields the same `NameError: name 'deck_clearance' is not defined` error.

SyntaxColoring, 2020-02-28:
Oh. This is fun.

We execute protocols like this:

https://github.com/Opentrons/opentrons/blob/f99a5f8b42941ba89bed8f2282ae02ecf6786f75/api/src/opentrons/protocol_api/execute.py#L92-L94

Which triggers this [`exec()`](https://docs.python.org/3/library/functions.html#exec) behavior:

> ...If exec gets two separate objects as globals and locals, the code will be executed as if it were embedded in a class definition.

In plain Python, if you embed a list comprehension in a class definition, it won't work [because of reasons](https://stackoverflow.com/questions/13905741/accessing-class-variables-from-a-list-comprehension-in-the-class-definition). This is why my original example raises an error.

```python
class C:
    x = 1
    y = [x + i for i in range(5)]
    # NameError: name 'x' is not defined
```

There's [similar weirdness](https://stackoverflow.com/questions/2024566/access-outer-class-from-inner-class-in-python) for embedding a *class* inside a class definition. This, I think, is what @Maxpyt was talking about.

```python
class C1:
    def f():
        return 123
    class C2:
        x = f()
        # NameError: name 'f' is not defined
```

It looks like the right thing to do is to provide only a `globals` to `exec()`:

> If only `globals` is provided, it must be a dictionary...which will be used for both the global and the local variables. [...] Remember that at module level, globals and locals are the same dictionary.

PR #5292 (2020-03-25): fix(api): exec protocol contents in module mode
if you give exec() separate locals and globals, it executes the provided
data as if embedded in a class definition which we very much do not want
- we want it to be executed as if in just a file, which of course it is.
This makes scoping work the same as it does when people write their
protocols and expect them to work.

While we're at it, call the argument we feed to the protocol's run
method something a little less likely to collide.

Closes #4982
Closes #4981

## Risk assessment

Should be minor - narrowly scoped change and easy to check since [... cut]
commit 0174ab39 (2020-03-25): fix(api): exec protocol contents in module mode (#5292)

if you give exec() separate locals and globals, it executes the provided
data as if embedded in a class definition which we very much do not want
- we want it to be executed as if in just a file, which of course it is.
This makes scoping work the same as it does when people write their
protocols and expect them to work.

While we're at it, call the argument we feed to the protocol's run
method something a little less likely to collide.

Closes #4982
Closes #4981
closed 2020-03-25 by sfoster1