Opentrons/opentrons #4982: Protocols can access variables that they don't define

SyntaxColoring, 2020-02-12:
Well, at least one variable:

```
metadata = {"apiLevel": "2.1"}

def run(protocol):
    # Note `context` is never defined here.
    protocol.comment(str(protocol is context))
```

```
$ opentrons_simulate test.py
...
True
```

I'm assuming an [`eval()`](https://docs.python.org/3/library/functions.html#eval) somewhere is forgetting to pass `globals` and `locals`, or something.

Possibly related to #4981.

sfoster1, 2020-03-25:
I think the farthest we'll go with this is to rename that variable to something a little less likely to collide (e.g. `__context`). To execute the run function properly we need to do it in an eval so we can control the local environment, and to do that we need to provide the function argument, so it needs to be bound in the local environment somewhere.

SyntaxColoring, 2020-03-25:
That seems fine.

Out of curiosity, why can't we just execute the protocol like:

```python
with open('uploaded_protocol_file.py', 'w') as f:
    f.write(protocol_source)
import uploaded_protocol_file
uploaded_protocol_file.run(context)
```

That would also make `metadata` inspection easier, right?

PR #5292 (2020-03-25): fix(api): exec protocol contents in module mode
if you give exec() separate locals and globals, it executes the provided
data as if embedded in a class definition which we very much do not want
- we want it to be executed as if in just a file, which of course it is.
This makes scoping work the same as it does when people write their
protocols and expect them to work.

While we're at it, call the argument we feed to the protocol's run
method something a little less likely to collide.

Closes #4982
Closes #4981

## Risk assessment

Should be minor - narrowly scoped change and easy to check since [... cut]
commit 0174ab39 (2020-03-25): fix(api): exec protocol contents in module mode (#5292)

if you give exec() separate locals and globals, it executes the provided
data as if embedded in a class definition which we very much do not want
- we want it to be executed as if in just a file, which of course it is.
This makes scoping work the same as it does when people write their
protocols and expect them to work.

While we're at it, call the argument we feed to the protocol's run
method something a little less likely to collide.

Closes #4982
Closes #4981
closed 2020-03-25 by sfoster1