Opentrons/opentrons #4991: Upgrade Python dev environment to 3.7+

iansolano, 2020-02-13:
## overview

We need to upgrade python in our dev environment to 3.7+ in order to support pydantic generics which come with the addition of Fast API.

amitlissack, 2020-02-14:
`pytest` is not exiting when using python3.7. It's because the `SynchronousAdapter` threads are not terminating. Why it works on python3.6 is baffling to me. 

If I call `join` on the `SynchronousAdapter` instances then pytest finishes and all tests pass, but getting access to these instances requires some pretty deep hackery on my part. 

I've spent most of today on this and am thinking it might not be worth the effort. Much of this code is on the path to refactoring. Additionally, we only really need python3.7 for the standalone API application.  What you all think?

mcous, 2020-02-14:
Sounds (to me, a non expert) a little like what was described in #4666

amitlissack, 2020-02-14:
> Sounds (to me, a non expert) a little like what was described in #4666

It is exactly the same issue.

sfoster1, 2020-02-18:
Yeah I think it has to do with process termination behavior of the interpreter and, like you've probably noticed, it was always incorrect code - 3.7 is just stricter about it. We can sit down about it sometime - I think that the problem is we need much more explicit lifetimes of the adapters.

Unfortunately I don't think #4870 will help in the tests - it just moves the problem - but instead we should be able to explicitly stop and terminate the worker threads in pytest fixture cleanups.

commit 8d29ef63 (2020-02-28): chore(api): Upgrade api to python 3.7 (#4999)

* Upgrade api to python 3.7

* update api docs to indicate we are going from py3.6.4 to py3.7.6

* change contributing to indicate 3.7.6 of python

closes #4991
closed 2020-02-28 by amitlissack