Opentrons/opentrons #5035: Deck Calibration Check: Tiprack Calibration Check

umbhau, 2020-02-20:
As an OT-2 user testing my robot’s calibration, I would like to be informed if the calibration data implied by my calibrated tiprack position is bad enough to immediately warrant a re-calibration, so that I do not waste time (or risk a crash) by proceed ingwith the rest of the calibration check.

## Acceptance Criteria
-  [ ] If user attempts to initiate tip pickup while the nozzle’s nominal distance from flush with the top-center of its target tip is greater than the specified maximum (see below), display a ‘Bad Deck Calibration’ modal.
-  [ ] New ‘Bad Deck Calibration’ modal informs user that their specified tip position is significantly off from the expected location of the [target tip] position for a standard Opentrons tip in [tip rack slot]
-  [ ] Modal informs user that they should confirm they are using the correct Opentrons-brand tips
-  [ ] Modal informs user that they should perform deck calibration 
-  [ ] User has option to cancel and can exit Deck Calibration Check flow and return to Robot page

Acceptable difference between nominal and user-specified tip position
- P10, P20, P300: +/- 5mm in X, Y, or Z
- P1000: +/- 10mm in X, Y, or Z

pantslakz, 2020-02-25:
TBD: Does modal trigger "bad calibration data" alert on robot card (#5044)

b-cooper, 2020-05-26:
closed by #5499 and #5583

closed 2020-05-26 by b-cooper