Opentrons/opentrons #5038: Deck Calibration Check: Tip Pickup and Calibration Data Pre-Check (2nd Pipette)

umbhau, 2020-02-20:
As a user, I would like to be guided through picking up a tip with my second pipette, if any, so that I can use it to check my mount offset. 

I would like the robot to validate that the tip position is close to its expected position for a 'good' calibration before proceeding, to prevent downstream crashes and wasted effort. 

## Acceptance Criteria
### Tip Pickup
- [ ] If second pipette mounted, guide user through picking up tip second pipette
  - [ ] If second tiprack used (for different tip type), tiprack should be in slot 8, per #5033 
- [ ] Pipette automatically moves to a safe distance from top of pipette tip, ignoring any tiprack labware calibration
- [ ] User prompted to jog to correct tip pickup position
  - [ ] IF single channel and using new tip rack, user guided to A1 tip position
  - [ ] IF single channel and using same tip rack as 1st pipette, user guided to B1 tip position
  - [ ] IF multi channel and using new tip rack, user guided to H1 tip position with rear pipette nozzle
  - [ ] IF multi channel and same tip rack as 1st pipette, user guided to G1 tip position with rear pipette nozzle
- [ ] User provided with visual guidance and standard jog controls
-  [ ] User prompted to initiate tip pickup
- [ ] Pipette picks up tip using standard tip pickup method
- [ ] User prompted to confirm tip pickup was successful
- [ ] IF tip pickup fails, prompt user to re-try
- [ ] IF tip pickup successful, check tip pickup position (see below)

### Check Tip Pickup Position
- [ ] Validate that user's specified tip pickup position is close to the expected location of the tip, given the current deck calibration (but ignoring the tiprack's labware calibration)
  - [ ] Range of 'close enough' distances specified in 'resources' below
- [ ] If tip pickup position is outside of range, display ‘Bad Deck Calibration’ modal from #5035 
- [ ] If tip pickup position is inside range, proceed to 'Calibration Check Successful' modal

pantslakz, 2020-03-02:
If using same tip rack - will pick up from B1 instead of A1.

b-cooper, 2020-05-26:
closed by #5499 and #5583

closed 2020-05-26 by b-cooper