Opentrons/opentrons #5097: bug: Log download button may save a .tmp file instead of api.log and serial.log

SyntaxColoring, 2020-02-27:
One customer reported that the log download button prompted them to save a single `e1b6579d-029b-48cd-9c7a-2c7cf97b0143.tmp` file, instead of the expected `api.log` and `serial.log`.

The .tmp file looks like a serial log. I can't tell if it's been truncated or not.

amitlissack, 2021-02-12:
@SyntaxColoring this bug is a year old. Should we devote time to it?

SyntaxColoring, 2021-02-12:
I think it's probably worth a time-boxed reading of the code (on both the robot and app side) to see if anything sticks out as a likely culprit.

If not, I think we can close this as unable-to-reproduce.

SyntaxColoring, 2021-02-19:
I'm closing this since:

* We haven't been able to reproduce it.
* After reading the code for a couple of hours, I couldn't find any likely culprit.

In the code, I checked:

* `app/` and `app-shell/`, which don't do much other than delegate to `electron-dl` to download the files. They don't do any of their own file management or .tmp stuff. So, likely not their fault.
* `electron-dl`'s source, which doesn't do much other than delegate again to Electron to download the file. Again, no file management or .tmp stuff.
* `opentrons.system.log_control`, which returns a `bytes` instead of a named file, so the .tmp couldn't be its fault.
* `robot_server.service.legacy.routers.logs`, which just passes along those `bytes`, and never explicitly sets a filename.

¯\\\_(ツ)\_/¯

closed 2021-02-19 by SyntaxColoring