Opentrons/opentrons #5170: feat: Blow out distribute() disposal volume in places other than the fixed trash

SyntaxColoring, 2020-03-05:
# Background

`distribute()` has a `disposal_volume` parameter that sets the minimum amount of liquid to be kept in the tip at any given time. The disposal volume is blown out into the trash before going back to the source for more liquid.

# Problem

This wastes reagent. We trash the disposal volume, but in many cases, we could blow it back into the source well to be reused.

The existing behavior makes sense when we're not concerned about destinations contaminating each other (we're using a `distribute()`, after all), but we are concerned about the destinations contaminating the source (maybe because the source will later be used for other things). I'm not sure if this case is common enough for it to be the default.

# Request

Allow a `distribute()`'s disposal volume to be blown out in the source well, instead of the trash.

Perhaps also allow it to be blown out to any arbitrary location.

SyntaxColoring, 2020-03-10:
You can *kind of* do this by changing `pipette.trash_container`, but that has two problems:

* It conflates solid (tip) trash with liquid (reagent blow out) trash.
* It accepts a labware object and assumes well A1; you can't change which well.

PaulBisesi, 2020-06-24:
As a user, this feature would be very useful. We achieved this behavior in our apiv1 protocols by passing a location to a `distribute()`'s  `blow_out` parameter. In apiv2 it seems this functionality no longer works, as the command will ignore anything in the `blow_out` parameter (including `False` or a location) and blow out in the trash.

SyntaxColoring, 2020-06-24:
Possible duplicate of #4940 (haven't read very closely to see if there are any differences, yet).

SyntaxColoring, 2020-07-27:
From #5734:

> When using the aspirate / dispense commands in python you can access the source well for blow out. This is vital when using valuable reagents. However, this cannot be accessed in the distribute/consolidate commands which are necessary for our protocol.

alexjoel42, 2020-07-29:
Hi! 
While this doesn't resolve the issue, it is a useful example of how to use the blow_out() functionality using aspirate() and dispense() commands to blow_out() places other than the trash.  
`
from opentrons import protocol_api
metadata = {'apiLevel': '2.4',
            'protocolName': 'fix blow out',
            'author': 'N/a',
            "description': 'This is a protocol to get around the .distribute command's lack of blow_out specificity "}
            
def run(protocol: protocol_api.ProtocolContext):
    tiprack_1 = protocol.load_labware('opentrons_96_tiprack_20ul',location = '1')
    # the big one
    reservoir = protocol.load_labware('usascientific_12_reservoir_22ml',location = '2')
    well_plate = protocol.load_labware('nest_96_wellplate_100ul_pcr_full_skirt','3')
    p20 = protocol.load_instrument('p20_multi_gen2','right', tip_racks=[tiprack_1])
    p20.pick_up_tip()
    p20.aspirate(20,reservoir.wells()[0].bottom())
    for i in range(0,3):
        p20.dispense(5,well_plate.wells()[i]).blow_out()
    p20.drop_tip()
`

pou-pou, 2020-08-18:
I would be needing that functionality too, but I ended up doing a version for myself. The multidispensing management was particularly tricky. Perhaps this can help anyone doing the same thing:

[distribution.txt](https://github.com/Opentrons/opentrons/files/5091843/distribution.txt)

The function 'distribution' accepts a list of wells, the volume to be dispensed per well and the maximum volume contained by the pipette and returns the list of trips the pipette makes with the wells that will be used for each of the trips.

James-Kitson, 2021-05-14:
> As a user, this feature would be very useful. We achieved this behavior in our apiv1 protocols by passing a location to a `distribute()`'s `blow_out` parameter. In apiv2 it seems this functionality [...]

This is especially useful if you're setting up tagged PCRs for next gen sequencing. In this situation, I want to distribute my tagged primers to multiple plates then drop tips without any blowout to prevent any tagged primer aerosols from floating around in the robot.

SyntaxColoring, 2021-05-14:
>  I want to distribute my tagged primers to multiple plates then drop tips without any blowout to prevent any tagged primer aerosols from floating around in the robot.

@James-Kitson Unless I misunderstand, that might be a separate problem. When our pipettes drop tips, they have to blow out *somewhere,* at *some point,* because the ejector is controlled by the same motor that controls the plunger. Ejecting a tip is like blowing out really far.

James-Kitson, 2021-05-14:
> @James-Kitson Unless I misunderstand, that might be a separate problem. When our pipettes drop tips, they have to blow out, because the ejector is controlled by the same motor that controls the plunger. Ejecting a tip is like blowing out really far.

Ah yes fair enough, maybe I just need to play with carry-over volumes then!

mattwelch, 2021-06-03:
Bug triage results: MMM