Opentrons/opentrons #5171: bug: Simulations happen in real time

open3dgenome, 2020-03-06:
I am setting up a protocol where I need to mix a sample for 30 min. Tests without the time restriction work fine but it is cumbersome that when the real case scenario is needed (`elapsed < 30*60`) the simulation of the protocol on the app takes 30 mins to run.

Is there a way to speed up simulations? or what is the recommended approach for a case like this?

The relevant part of the protocol looks like this:
```
    start_time = time.time()
    end_time = time.time()
    elapsed = end_time - start_time
    total_cycles= 0
    pipette.move_to(sample_bot)
    while(elapsed < (30*60)):
        total_cycles += 1
        end_time = time.time()
        elapsed = end_time - start_time
        pipette.aspirate(volume=mix_vol, location=sample_bot, rate =0.5)
        for i in range(repetitions):
            pipette.dispense(volume=mix_vol-1, location=sample_bot, rate =0.5)
            pipette.aspirate(volume=mix_vol-1, location=sample_bot, rate =0.5)
        pipette.dispense(volume=mix_vol, location=sample_bot, rate =0.5)
```

open3dgenome, 2020-03-06:
As an appendix to the issue. The real problem of this is not that is cumbersome but that (maybe due to a timeout lock) the simulation never succeeds and actually blocks the communication between the app and the robot. So the protocol is impossible to run.

mcous, 2020-03-06:
@open3dgenome I would not recommend a busy `while` loop for this; like you said it's going to lock all sorts of stuff up.

Are you able to use [`ProtocolContext.delay`](https://docs.opentrons.com/v2/new_protocol_api.html#opentrons.protocol_api.contexts.ProtocolContext.delay) for this? The delay will be skipped during simulation and only occur during run.

Something to keep in mind when using `time.time()` in Python protocols: the protocol is executed twice independently: Once for simulation and then once again during run. The App UI is populated via the results of simulation, so if you try to log out the results of `time` calls (perhaps via [`ProtocolContext.comment`](https://docs.opentrons.com/v2/new_protocol_api.html#opentrons.protocol_api.contexts.ProtocolContext.comment), the logs you see in the app will _not_ reflect what actually happens during the run, but the run will still function with the proper delays

open3dgenome, 2020-03-10:
Hi @mcous, thanks for your answer. It clears things up.

Using the delay is a good solution for the current protocol I am developing, but it is not precise.

How would you implement a loop that would last for a given period of time when using libraries such as `time.time()` is not possible?

mcous, 2020-03-25:
> Using the delay is a good solution for the current protocol I am developing, but it is not precise.

Could you elaborate on what you mean by this? ProtocolContext.delay simply waits for an [asyncio.sleep]() to complete before proceeding, but skips the check during run. It takes a float, so it should wait as precisely as you tell it to

> How would you implement a loop that would last for a given period of time when using libraries such as time.time() is not possible?

I might be able to give a more specific answer if I understood what sort of time-based (rather than volume or repetition number based) flow you were attempting to implement, but generally you can continue to use `time.time()` as long as you wrap any calls in a check to [ProtocolContext.is_simulating](https://docs.opentrons.com/v2/new_protocol_api.html?highlight=is_simulating#opentrons.protocol_api.contexts.ProtocolContext.is_simulating) to ensure that you *don't* do any busy waits during simulation.

This will certainly result in an incorrect run log display in the app because it's not (yet) set up to handle a discrepancy between simulation and run.

I'm going to close this ticket because the simulation system is functioning as expected, but feel free to continue posting in this thread and I (or one of my colleagues) will do our best to answer!

closed 2020-03-25 by mcous

calebwin, 2020-11-12:
@mcous Is there a way to make the _opposite_ happen? In particular, is there a way to ensure that all calls to `asyncio.sleep` in `delay` as well as in the implementations of the various modules - _do_ block for the whole time, all while simulating using an `opentrons.simulate.get_protocol_api("2.7")`?