Opentrons/opentrons #5200: Discovery: Spike: Address' stability

sanni-t, 2020-03-10:
This ticket discusses stability of the robot's MAC addresses, ipv4 and ipv6 addresses
Deliverables:
- [ ] There seems to be cases where the robot's MAC address is unstable. Investigate if this is true and whether this is expected & harmless or is causing trouble. In case it's trouble, how do we fix it?
- [ ] What is going on with ipv6 addressing? Why does an ipv6 addr remain reachable when ipv4 disappears?
- [ ] What if we made link local address static by default?

mcous, 2020-03-25:
I am able to reliably reproduce IPv4 failures with a 3.16.1 RPi and my home macOS desktop. It helps to have the Pi on Wi-Fi with an SSH terminal in via the Wi-Fi IP

1. Connect Pi to desktop directly with an Ethernet cable
    - Observer NetworkManager brings up `wired-linklocal` connection with IPv4 and IPv6 addresses
2. Tell the desktop to go to sleep
3. After several seconds, `systemd-networkd` will issue a carrier down followed by a carrier up
    - NM will, accordingly, take the connection down and then up again
    - When the connection comes back up, it will have an IPv6 address but no IPv4 address
4. Wake the desktop back up
5. Carrier will go down and then up again, along with the connection
6. `wired-linklocal` connection will be **up without an IPv4 address and with an IPv6 address that the desktop cannot route to (I could be wrong here but I was unable to resolve ICMPs to the self-assigned IPv6)

The only way I can get NM to take down `wired-linklocal` (which it thinks is successfully up) is to either:

- Fully unplug the Ethernet for about 30 seconds and then plug it back in
- Issue an `nmcli connection down wired-linklocal` followed by `nmcli connection up`

Next steps for this particular failure mode:

- Try to figure out if there are any underlying kernal / systemd bugs going on here
- Try to figure out if there's a way to configure NM to require an IPv4 to call `wired-linklocal` up
- Investigate `avahi-autoipd` as a potential replacement for self-assigning link-local IPv4 addresses
    - **Note**: `avahi-autoipd` was already ruled out once as a solution because of weirdness with how it interacts with networkd and NetworkManager
    - See Opentrons/buildroot#9 for more details

mcous, 2020-04-28:
Needed items for closing out this research ticket:

- [ ] What can we do to make Wi-Fi MAC address stable?
- [ ] Collect remaining logs for IPv6 assignment without IPv4
- [ ] Evaluate feasibility of stable link-local IPv4
    - Tentatively, this may not be worth the effort
    - This has implications for how the Discovery Client caching logic

SyntaxColoring, 2020-04-28:
> Evaluate feasibility of stable link-local IPv4

It probably won't make sense once OT-2s expose Ethernet ports, and people can easily plug them into unmanaged switches.

nickcrider, 2020-05-08:
Customer reported this workaround to achieve stable MAC address on the Pi WLAN adapter:

1. SSH to `/etc/NetworkManager/conf.d`
2. Remounted `/` to gain write access `mount -o remount, rw /`
3. Create `wifi_rand_mac.conf` in the directory with the following contents
```
[device]
wifi.scan-rand-mac-address=no
```
4. Reboot

After this they reported the MAC address was stable and has the correct beginning format for a Raspberry Pi (e.g. `B8:27:EB…`)

nickcrider, 2020-05-11:
Hey so I followed up again with the user on this. Looks like a random MAC per SSID is a [default setting in ArchLinux](https://wiki.archlinux.org/index.php/NetworkManager)

> Randomization during Wi-Fi scanning is enabled by default, but it may be disabled by adding the following lines to `/etc/NetworkManager/NetworkManager.conf` or a dedicated configuration file under `/etc/NetworkManager/conf.d`

Seems like the user followed exactly this process. Hopefully this is a pretty quick fix and tagging @sfoster1 as I think this is his domain of expertise.

mcous, 2020-05-12:
Wi-Fi MAC address stability tracking ticket: #4758
IP address stabillity / reliability epic: #5652

closed 2020-05-12 by mcous