Opentrons/opentrons #5213: bug: Confusing error message when pick_up_tip is used on a bare location (minor)

theosanderson, 2020-03-12:
# Overview

This is a very minor low-priority issue where the error message isn't quite as expected for a niche usage.

# Current behavior

If you say 

```
pipette.pick_up_tip( opentrons.types.Location(opentrons.types.Point(x,y,z),"my_location"))
```

You get the suboptimal error:
```
  File "/home/pi/bare/lib/python3.6/site-packages/opentrons/protocol_api/contexts.py", line 1267, in pick_up_tip
    assert tiprack.is_tiprack, "{} is not a tiprack".format(str(tiprack))
UnboundLocalError: local variable 'tiprack' referenced before assignment
```

This is because this unexpected usage falls through a crack in the if-then logic whereby the location is a location but is neither a well nor labware, and so the tiprack variable is not set. https://github.com/Opentrons/opentrons/blob/edge/api/src/opentrons/protocol_api/instrument_context.py#L634

# Expected behavior
Perhaps a clearer error message saying that bare locations are not supported. Ideally for my use-case this would actually be supported, but I understand that is niche and I'm now figuring out how to work around but I thought I'd log the variable-used-before assignment issue.

mcous, 2021-12-10:
No longer appears to be an issue after reading the current code

closed 2021-12-10 by mcous

theosanderson, 2022-10-09:
Fwiw it's still an issue, but may well not be worth your time
[image]