Opentrons/opentrons #5240: bug: "Connection failed" dialog is taken away as a robot times out of the list

SyntaxColoring, 2020-03-17:
# Overview

When you try to connect to a robot over a broken connection, two things happen:

1. The act of connecting will time out, eventually spawning a "Connection failed" modal dialog.
2. Separately, the Opentrons App will notice that it hasn't heard anything from the robot in a while, so it will remove it from the list. When it removes the robot from the list, it also removes any dialogs that its page spawned – including the above.

These two things race against each other, and their timeouts are of similar duration. So, the user may see a brief flash of "Connection failed" before the robot page is wiped away.

Although I haven't tested it, I assume this also happens for anything else that takes a network request – turning on the deck lights, homing the robot, whatever.

# Steps to reproduce

1. Have a robot ready to connect to in the Opentrons App.
2. Break the connection to that robot.
3. Immediately try to connect to that robot.
4. Wait. A "Connection failed" dialog will appear.
5. Wait some more. The robot will be removed from the list because the app hasn't heard from it in a while. Simultaneously, the "Connection failed" dialog will go away.

# Expected behavior

First, I'm not sure if a modal dialog is the best UX for a connection failure. But, if we take that for granted, then I *think* the modal should stick around until the user dismisses it, so they have time to read what it says.

mcous, 2020-03-18:
## TL;DR: solution notes

- Do not re-direct route if robot is not connectable; keep robot name in route and display splash in robot page component
    - This should ensure any required state (the robot name) for displaying alerts remains in place, so modals don't disappear along with the contents of the page

Implementation / fix notes:

> 1. You try to connect to a robot over a flaky connection.

User has clicked the "connect" button or toggle, which will cause the app to attempt to open a WebSocket for the RPC server

> 2. You may or may not see a "Could not connect to robot" dialog.

User will see this if the WebSocket fails to open or if the RPC server fails to negotiate the initiate the connection properly

> 3. The robot is removed from the list, and you're kicked out of its page. If you saw a "Could not connect to robot," it's wiped away.

As noted, this is independent of the WebSocket stuff. We don't have UI for a robot page for a robot that is not connectable. I think this would be a good thing to have for a variety of reasons, so I'll likely open another ticket and drop it here

> 4. The connection is fixed. (The microwave in the next room turns off, the USB cable is plugged back in, whatever.)

**Important to note**: referring here to the network connection, not the specific RPC WebSocket

> 5. You reconnect to the robot and immediately see a "Could not connect to robot" dialog. The dialog is stale, referring to when you lost connection before. But the way it appears makes it look like it's referring to the connection that you just regained.

**Also important to note:** When this happens, the user **has not reconnected the RPC WebSocket**. The **networking** connection has become good enough again to allow HTTP `GET /health` to succeed, so the robot page is displayed again. Certainly the user thinks they have "reconnected", but at this point the app has not established a persistent WebSocket connection with the robot.

So, in a technical sense, the "could not connect to robot" dialog is not stale here. The last time the user tried to establish an RPC WebSocket connection to this robot, it failed, and the user never explicitly dismissed that dialog, so it displays.

> Although I haven't tested it, I assume this also happens for other interactions that take a network request, like turning on the deck lights or homing the robot.

These interactions are different, because they happen over HTTP requests and do not involve the persistent RPC WebSocket. However, if the HTTP request goes out and fails, that error message will still be present if the robot drops off the list and comes back.

mcous, 2020-09-21:
As of 3.20.0, the app will not remove any robot from the UI that it's been talking to while the app has been open. Consequently, the "Could not connect" modal will not disappear until the user clicks the close button

closed 2020-09-21 by mcous