Opentrons/opentrons #5241: bug: Multiple connection failures each spawn their own "could not connect to robot" dialog

SyntaxColoring, 2020-03-17:
# Current behavior

When a network request like connecting to the robot times out, it spawns a "Connection failed" dialog.

If multiple requests time out – which can happen if a user impatiently clicks a button a bunch of times – each one will spawn its own dialog. So, once you close the first one, the next one will appear shortly to tell you something you already know. This can get fairly spammy and confusing.

# Steps to reproduce

Issue #5240 makes this a little tricky to reproduce. You have to get lucky with the timing, or the dialogs will be hidden before you have the chance to click through them. But, basically:

1. Have a robot in the list ready to connect to.
2. Break the connection to it.
3. Immediately, click the connection switch or button a few times.
4. Wait for the "Connection failed" dialog to appear, then immediately exit it. A second one will appear shortly after.

# Expected behavior

There are probably a few ways of fixing this.

* Maybe I'm not allowed to spam-click any buttons. If I do something that requires network activity, I'm not allowed to try that thing again until it either succeeds, fails, or times out.
* Maybe the errors aren't modal dialogs. Maybe they're inline error messages instead. Then, it wouldn't matter if they keep coming in, because they're less obtrusive.
* Maybe network failure messages are coalesced. If I just saw one, I don't need to see another.

mcous, 2020-03-18:
> When a network request like connecting to the robot times out, it spawns a "Connection failed" dialog.

Slight correction: when _the_ request to establish an RPC websocket connection fails, the app shows a Connect failed dialog. It is specific to establishing the RPC connection and not a generic network failure error. Pretty straightforward to ignore any connection requests if one is in-flight if this is a high-priority bug

SyntaxColoring, 2020-03-19:
I don't think this ticket is particularly high-priority. Just one of many little low-priority things that contribute to a high-priority area of pain.

commit 9465cef3 (2020-04-07): fix(app): ensure only one RPC connect request can go out at once (#5322)

Closes #5241, closes #5307
closed 2020-04-07 by mcous