Opentrons/opentrons #5243: bug: Robot discovery duration may be too short to realistically succeed on the first try

SyntaxColoring, 2020-03-17:
# Overview

I find myself often having to click the "Try again" button at least 2 or 3 times before a just-plugged-in robot shows up. I don't think there's anything wrong with the connection; I think the app is just giving up too early.

This may make the system appear flakier than it actually is.

# Details

On my MacBook, after I plug in the USB cable, it takes something like 25 seconds for a link-local IP to be assigned. The robot discovery period seems to be 30 seconds, leaving only a 5-second window. And I'm unsure of how densely the app sends out mDNS requests in that window.

We can assume that users won't appreciate that it takes time for the network connection to be ready. They will click "Try again" as soon as they plug the USB cable into their computer. The robot discovery period should account for this.

# Expected behavior

The "Try again" or "Refresh list" button should work the first time, when I:

* Click it immediately after plugging in a USB cable to a robot that's ready.
* Or, when I click it immediately after the blue light on the robot turns solid.

# To do

* Research how long after plugging in the cable it usually takes for a computer to assign itself a link-local IP, on Windows 10, macOS, and Ubuntu.
* Research how long after turning an OT-2 on or how long after its blue light turns solid it usually takes for it to be ready to respond to mDNS requests.
* Confirm that the robots that there's nothing different here between the robots that we use in-house and the robots that customers are receiving. i.e. make sure there's nothing left over from weird factory provisioning that could delay the OT-2s' network readiness.

mcous, 2020-03-17:
The problem here isn't the density of mDNS requests, or that the discovery period is too short (because the app never stops discovering, it just slows down its `GET /health` HTTP polling interval). The problem here is that the mDNS browser only checks the computer's interface list for binding at the beginning of any given discovery period, and won't refresh that list until a new fast discovery period is started (for example, by clicking the button).

Increasing the timeout wouldn't help, because the interface would just come up during that longer period without getting attached to the mDNS browser. The fix here is to automatically tear down and re-create the browser (like happens when the "refresh" button is clicked) when the app is able to detect a network interface change of the computer.

I've got this fix slated for a ticket to be written today as part of the u2e driver update project if that sounds good to you

SyntaxColoring, 2020-03-18:
Aha.

Yeah, once that ticket is up, we can close this one in favor of it.

SyntaxColoring, 2020-04-01:
Closing this in favor of #5343, per above.

closed 2020-04-01 by SyntaxColoring