Opentrons/opentrons #5279: bug: Labware Creator doesn't import custom API load names

SyntaxColoring, 2020-03-24:
# Current behavior

When you import a custom labware definition to edit it, it assumes that the API load name was unchanged from the default. If the API load name was customized when the definition was originally created and exported, that customization isn't imported.

I haven't tested the display name.

# Steps to reproduce

1. Create a labware definition with a customized display name or API load name.
2. Import that new labware definition.
3. Note that "API Load Name" field is pre-populated with the default API load name, not the one you set.
4. Export the definition.
5. Open the .json file and note that the load name has changed back to the default.

For example, try [Applied Biosystems 96 Aluminum Block 200 µL.json](https://github.com/Opentrons/opentrons/files/4377923/Applied.Biosystems.96.Aluminum.Block.200.L.json.zip), which should have a load name of `ab_96_aluminumblock`, but reverts to `appliedbiosystems_96_aluminumblock_200ul`.

# Expected behavior

The "Display Name" and "API Load Name" fields should be pre-populated with the values from the .json file.

pantslakz, 2020-03-25:
@SyntaxColoring I believe this is not a bug - this was an intentional behaviour we implemented to simplify things.

Reasoning behind this:
- if user exports a labware, finds an error (ie. volume is incorrect), re-uploads it to labware creator and makes an edit - what should happen to the API load name? Should the load name reflect the new volume? 
- similarly, what happens if a user wishes to use an existing labware as template for a new one?

We decided to revert to the default load name, as this new definition is geometrically different and we err on the safer side by having the name align with the updated inputs within the fields.

It's up to the user to change the API load name if they wish to use something custom.

@IanLondon @howisthisnamenottakenyet lmk if you disagree or if I missed anything.

SyntaxColoring, 2020-03-28:
That makes sense to me.

I think we can close this ticket, for now. If it confuses anyone again, we'll reopen it and ponder it some more.

Labware versioning could also affect our decision, if/when we implement it.

closed 2020-03-28 by SyntaxColoring