Opentrons/opentrons #5307: bug: Connecting to a robot behaves confusingly when connections are non-instantaneous

SyntaxColoring, 2020-03-28:
# Overview

On this network, it's taking about 2 seconds from clicking the switch next to a robot's name to the connection succeeding. Once connected, flipping the deck lights on and off happens instantaneously, so I'm not sure where the latency is coming from.

The mysterious latency is one problem. More importantly, though, it seems to open the door to some other confusing behaviors.

To summarize:

* Establishing a connection has an unexpectedly high latency.
* There's no loading spinner when I click the connection switch.
* If I click the connection switch more than once before it succeeds, the app behaves as if the connection were canceled?
* Such a cancelation apparently puts the app UI into a weird state where it sort of thinks I'm connected and sort of doesn't.

# Steps to reproduce / current behavior

If I:

1. Click the connection switch
2. Quickly click the connection switch again (because I'm an impatient user and there's no spinner to give me feedback)

I get kicked to this screen:

[image]

* The Protocol, Calibrate, and Run tabs are all immediately grayed out. The app has kicked me to this screen as if the connection succeeded, but now it's behaving as if I don't have a connection.
* I don't get any "Connection to robot lost" dialog.
* If I click the "Reset run" button, I get an infinite spinner.

Then, if I:

3. Go back to the Robots tab

I see this, as if I had never been connected at all:

[image]

On the other hand, if I'm patient after step 1 and only click the connection switch once, things behave sanely. I get kicked to the Run tab, but the "Reset run" button behaves normally; and, if I go back to the Robot tab, I see that I'm connected to that robot.

# Expected behavior

* Connecting to a robot probably shouldn't take 2 seconds if flipping the lights on and off is instantaneous? I don't know if this is expected.
* If connecting to a robot isn't instantaneous, I should see some kind of loading spinner, for immediate feedback.
* Clicking the connection switch or button more than once before the connection goes through shouldn't affect the outcome.

mcous, 2020-03-30:
> Connecting to a robot probably shouldn't take 2 seconds if flipping the lights on and off is instantaneous? I don't know if this is expected.

Fully expected. Clicking the connect button connects to the WebSocket RPC server. There's a handshake and serialization / deserialization of a lot of data. The RPC server is bad and we're replacing it, but that's outside the scope of this ticket

> If connecting to a robot isn't instantaneous, I should see some kind of loading spinner, for immediate feedback.

Easy. Can this ticket be closed if this is implemented?

> Clicking the connection switch or button more than once before the connection goes through shouldn't affect the outcome.

Is this covered by #5241?

SyntaxColoring, 2020-03-30:
> > Connecting to a robot probably shouldn't take 2 seconds if flipping the lights on and off is instantaneous? I don't know if this is expected.
> 
> Fully expected.

Great. I've removed that from this ticket.

> > If connecting to a robot isn't instantaneous, I should see some kind of loading spinner, for immediate feedback.
> 
> Easy. Can this ticket be closed if this is implemented?

The way I see it is: a spinner would filter out a certain kind of problematic user input. That would make the problem much more difficult to reproduce, but I don't know if it would cover _all_ the corner cases. And I guess I'm nervous that putting our defense against this problem in the UI code might make us vulnerable to a regression if the UI is ever redesigned or refactored.

But ultimately, if you think that a spinner would be a sufficient defense, then yes, that could close this ticket.

> > Clicking the connection switch or button more than once before the connection goes through shouldn't affect the outcome.
> 
> Is this covered by #5241?

Yes, I think your proposed solution in #5241 ("ignore any connection requests if one is in-flight") covers this, as well.

commit 9465cef3 (2020-04-07): fix(app): ensure only one RPC connect request can go out at once (#5322)

Closes #5241, closes #5307
closed 2020-04-07 by mcous