Opentrons/opentrons #5394: bug: PD Error for Pause Step remains with other steps selected

dozercodes, 2020-04-08:
# Overview

After adding a pause step that waits for a temperature to be reached before adding the necessary temperature step, an error correctly appears in the deck map. However the error remains in place on the deck map, despite moving on to other steps. The error should not be present when the user selects a different step

# Current behavior

Error remains in place after the user selects a different step.
[image]

# Steps to reproduce
1. Create a Protocol
2. Add pipette(s) and the temperature module
3. On the design page, add a pause step
4. Choose the pause until temperature is reached option
5. Enter any value within the acceptable range
6. Save the step
7. Observe the error message
8. Add another step
9. Observe the error message is still present

# Expected behavior
The error should only appear when the pause step in question is selected

IanLondon, 2020-12-02:
This is the expected behavior of timeline errors, not specific to this "missing temperature step" error. Any timeline errors will stop propagation of the timeline at that error-ing step, and all timeline errors are all shown no matter what your selection is (eg, running out of tips will always show). We could do something like "only show timeline errors for the selected step" (which would have wider consequences for PD outside of this "missing temperature step" error) or as #5392 suggests, make this error a warning.

howisthisnamenottakenyet, 2021-03-18:
@IanLondon @anthonywilleee Should we do what Ian suggested? I noticed this come up in batch edit mode and it's a bit more conspicuous there.

closed 2022-09-30 by bndo