Opentrons/opentrons #5406: bug: `import opentrons` is noisy

SyntaxColoring, 2020-04-10:
# Overview

`import opentrons` prints some noisy and meaningless-to-users stuff.

This shows up:
* In Jupyter.
* With `opentrons_simulate`.
* With standalone scripts using `opentrons.execute.get_protocol_api()` or `opentrons_simulate.get_protocol_api()`.

This is minor, but it has worried customers before. (They've thought that the "not found, loading defaults" message meant that something was wrong.)

# Current behavior

```shell
% python -c 'import opentrons'
Loading json containers...
Json container file load complete, listing database
Found 0 containers to add. Starting migration...
Database migration complete!
/Users/maxpm/.opentrons/deck_calibration.json not found. Loading defaults
/Users/maxpm/.opentrons/robot_settings.json not found. Loading defaults
%
```

The two "not found" messages at the bottom are printed to stderr.  The others are printed to stdout.

# Expected behavior

```shell
% python -c 'import opentrons'
%
```

sfoster1, 2020-04-20:
This actually will be fixed in 3.18 by https://github.com/Opentrons/opentrons/pull/5413 . Because we now load the robot lazily, the work that produces the messages won't happen until the robot itself is imported.

That said, the messages will still happen when the robot is imported.

SyntaxColoring, 2020-05-03:
It sounds like a fix for this has landed for APIv2, but the problem remains for APIv1. I don't think this is worth additional work to fix just for APIv1, so I'm retitling, tagging, and closing this ticket accordingly.

PR #5847 (2020-06-09): refactor(api): remove labware migration prints
This removes the labware database migration statements that are `print`ed every time the robot module is imported, which are confusing for end users, in favour of `log.info` messages. It leaves two print statements intact that are called when migration actually happens so that if the interface does not lock up without any information on progress. Changes welcome.

Largely fixes #5406.

## risk assessment

Low - no functional changes. These messages seem to date from balena.
closed 2020-05-03 by SyntaxColoring