Opentrons/opentrons #5428: bug: Canceling a protocol homes the pipette plungers, contaminating the pipette

SyntaxColoring, 2020-04-14:
# Overview

After hitting Cancel, the pipette plungers home along with the other axes, potentially aspirating liquid into the pipette stem and contaminating the pipette.

# Steps to reproduce

1. Attach a pipette you don't care about contaminating. (Or be ready to jump in and hit the kill switch.)
2. Run a protocol.
3. At a point when the pipette tip has liquid in it, hit Cancel.
4. The Z/A axes will home, then the X/Y axes, and finally the B/C axes. When the B/C axes home, they will pull the liquid in the tip up.

# Expected behavior

It should be safe to hit the Cancel button at any point in a protocol; it should never aspirate liquid further than it is already aspirated.

This could be implemented a few different ways, like:
* Blowing out liquid before homing the plungers, after a cancel.
* Or, ejecting tips before homing the plunger, after a cancel.
* Or, just not homing the plungers after a cancel.

SyntaxColoring, 2020-04-14:
This ticket is similar to #3738. However:

* That ticket is a feature request, while this ticket is a bug report.
* That ticket requests a specific remedy (eject tips before homing), while this 
* That ticket is about dangerous plunger homing behavior *in general,* while this ticket is specifically about what happens when you cancel a protocol.

James-Kitson, 2020-04-15:
Chipping in here as I had originally commented on [issue 3738](https://github.com/Opentrons/opentrons/issues/3738). I think you're right in general, there isn't really an issue with pipettes homing on the robot booting up, it's when you have to cancel a run that things get messy. One thing I'd think about is what happens to the already aspirated liquid in your implementations below. I'd suggest that either the liquid needs to be dispensed into the source well of the last command or the tips need to be ejected entirely into waste (probably safer overall). The third option below is probably less useful as it relies on the user remembering to manually remove tips for the setup to be made 'safe' before restarting. If that makes sense.

> This could be implemented a few different ways, like:
> 
> * Blowing out liquid before homing the plungers, after a cancel.
> * Or, ejecting tips before homing the plunger, after a cancel.
> * Or, just not homing the plungers after a cancel.

SyntaxColoring, 2020-04-21:
@James-Kitson 

Definitely. Avoiding the plunger home is only a partial solution. The full solution should get the robot back to a ready-to-run state, which probably means ejecting tips.

But: in case the full solution is hard to implement, I think the partial solution would still be a big improvement. It would at least let you stop the robot safely, so you have time to figure out what to do next.

PR #8229 (2021-08-30): fix(api): drop tip after cancel
Closes #7466
Closes #3738
Closes #5428

# Overview

This PR adds a functionality that if a protocol is cancelled mid-run and the pipette has a tip attached when the cancel is initiated, then the robot will empty the tip & drop it in the pipette's trash container before homing the plungers.

This is achieved in the following way:

Right after a cancel is issued:
1. pipette state is stored to determine whether it has a tip (and to save the last known plunger location but I’m not using that right now)
2. Cancel is fully executed with `home_after=False`. Hardware co [... cut]
closed 2021-08-30 by sanni-t