Opentrons/opentrons #5444: refactor(api): make interaction with API/ThreadManager more dev friendly

amitlissack, 2020-04-17:
Issues:
- `Controller`, `Simulator`, and `API` have the same interface yet no class relationship. These three classes need to be kept in synch.
- `ThreadManager` is the interface to the hardware. It wraps `API` but in doing so completely hides the interface. This causes a number of problems:
   - a need to ignore mypy errors on all calls to `ThreadManager` object.
   - The naming is really bizarre. 
   - Any developer using an IDE is hindered by there being no hints for calling `API` methods on a `ThreadManager` instance.

Proposal:
- `Controller` and `Simulator` and `API` all inherit from a common interface. Perhaps the common base class it the `Simulator`?
- Use mixin pattern to connect `ThreadManager` to `API`. `ThreadManager` can overload `build_hardware_simulator` and `build_hardware_controller` to capture managed object.

sfoster1, 2020-07-21:
The `Controller` and `Simulator` should definitely have an ABC or something but they are a different thing from the API. Similarly, we really want the threadmanager and synch adapter to wrap instances of the api rather than be is-a related.

amitlissack, 2020-07-23:
This is a really old ticket and sadly it's still hard to justify devoting time to it since the code works. 

I would update it now to say that we should prioritize getting rid of ThreadManager and SynchAdapter as the interface to the API. There is no way that a class with as complex an interface as API should be hidden behind something called a ThreadManager. 

I strongly feel that the pattern of wrapping a class and hiding its interface should be strictly avoided.  Users of IDEs and Mypy would appreciate it. Python's mixin pattern is a much better approach.  And even that should be used judiciously.

closed 2022-03-04 by mcous