Opentrons/opentrons #5447: bug: Despite good labware calibration, underlying calibration errors affect tip return height,

SyntaxColoring, 2020-04-19:
# Overview

Normally, labware calibration can make up for any errors in deck and pipette calibration.

Tip return is an exception. It's not enough to compensate with labware calibration. The underlying deck and pipette calibrations have to be correct, themselves.

Otherwise, when the pipette tries to return tips, it will go to the wrong height.

# Current behavior

The offset between the tip pickup position and the tip return position changes depending on the pipette's instrument offset. This is due to an inconsistency between tip pickup and tip return.

The tip *pickup* position relies on:

1. Deck calibration.
2. Labware calibration.
3. The x- and y-components of instrument offset.

The tip *return* position relies on all of that, too, but it *also* relies on:

4. The z-component of instrument offset (the tip length).

This inconsistency causes problems.

For one thing, it makes the tip return clearance vary from robot to robot, due to quirks in the robots' underlying deck and pipette calibrations.

It also makes tip return unnecessarily sensitive to calibration problems. For example, let's say:

1. We start with a deck calibration that's too high.  (Maybe because of issue #5022.)
2. We do pipette calibration.  Because the deck is calibrated too high, the tip length will come out too short.
3. We calibrate all our labware.
4. When the tip is returned at the end of labware calibration, the pipette will be too low. It will also be too low for tip returns in the protocol. Tip pickup, as well as movement to all the other labware, will work normally.

# Steps to reproduce

1. As a control experiment, upload, calibrate, and run tip_return_height_test.py.
    * Pay attention to the tip pickup and return height.
2. Reset the run.
3. Redo pipette calibration, but trigger the top switch a couple cm early.  (Trick the robot into thinking the tip is longer than it really is.)
4. Do labware calibration normally.
    1. The pipette will already be properly centered over tip A1; you won't have to make any adjustment.
    2. However, despite the tip rack's calibration looking good, the return height will be wrong.  It will be a couple cm too high.
5. Run the protocol.  The tip pickup and return heights will be the same as in labware calibration.

## tip_return_height_test.py

```python
metadata = {"apiLevel": "2.2"}
def run(protocol):
    tip_racks = [protocol.load_labware("opentrons_96_filtertiprack_200ul", s) for s in [1]]
    pipette = protocol.load_instrument("p300_single_gen2", mount="right") # To do: Test with a different pipette.
    
    for t in tip_racks:
        pipette.pick_up_tip(t["A1"])
        pipette.return_tip(t["A1"])
```

# Expected behavior

A pipette's tip return position should be a constant offset from its tip pickup position.

## Should both depend on tip length, or neither?

The important thing is that the tip pickup position uses the same calibration inputs as the tip return position.

This could be done by either changing tip pickup to depend on tip length, or changing tip return to not depend on tip length.  I'm not sure which is right.

On one hand, tip pickup, conceptually, has nothing to do with the tip length.  On the other hand, tip length calibration is effectively a compensation for deck calibration.

SyntaxColoring, 2020-04-22:
Per a conversation with @sfoster1, we don't plan on fixing this problem specifically. We think it will be a moot point after a planned calibration overhaul.

SyntaxColoring, 2020-04-22:
Until then: If tips are being returned at the wrong height for you, but the robot's positioning is otherwise correct, that indicates a problem with your underlying deck calibration or pipette calibration.

To fix it:
1. Factory reset your pipette calibration (**Advanced Settings** > **Factory Reset**). Doing this first works around an unrelated issue, #5022.
2. Redo deck calibration.

mcous, 2021-06-03:
This may be resolved by new Tip Length Cal flow. Double-check that it is before closing

closed 2022-09-30 by bndo