Opentrons/opentrons #5461: bug: Building Block Commands don't work with examples from the Docs

nickcrider, 2020-04-21:
# Overview

Building Block Commands don't work when used with example Accessor Methods in the APIv2 Documentation.

The vast majority users in the lab expect a multi-channel pipette to operate on an entire column (in a 96-well plate) at once, and when working with the multi are likely to use the given methods in the API documentation for accessing columns rather than wells. When used, these methods should produce the expected robot movement or a helpful error message.

# Current behavior

It seems to be a known issue that the building block commands only accept single well location as an input when working with the multi channel pipette. See: https://github.com/Opentrons/opentrons/issues/4946 However, this behavior is not mentioned in the documentation and is confusing to users who are accustomed to thinking about the multi-channel pipette as working on entire columns at a time.

So, when users try to access an entire column with the multi-channel pipette, as shown [here](https://docs.opentrons.com/v2/new_labware.html#accessing-groups-of-wells) in the docs by doing something like `p_multi.aspirate(10, plate.columns_by_name()['1'])` they receive a `TypeError` from our API rather than the expected movement.

# Steps to reproduce

Simulate the following protocol with version 3.16.1

```python
from opentrons import protocol_api

# metadata
metadata = {'apiLevel': '2.2'}

### Run Protocol ###
def run(protocol: protocol_api.ProtocolContext):

    # tipracks
    tiprack_multi = protocol.load_labware('opentrons_96_tiprack_300ul', '10')

    # labware
    plate_96well = protocol.load_labware('nest_96_wellplate_100ul_pcr_full_skirt', '1')

    # pipettes
    p_multi = protocol.load_instrument('p300_multi', 'left', tip_racks=[tiprack_multi])

    # commands

    p_multi.pick_up_tip()
    p_multi.aspirate(100, plate_96well.columns_by_name()['1'] )
    p_multi.dispense(100, plate_96well.columns_by_name()['10'] )
    p_multi.drop_tip()
```
Results in the following error:
```
opentrons.protocol_api.execute.ExceptionInProtocolError: TypeError [line 27]: location should be a Well or Location, but it is [A1 of NEST 96 Well Plate 100 µL PCR Full Skirt on 1, B1 of NEST 96 Well Plate 100 µL PCR Full Skirt on 1, C1 of NEST 96 Well Plate 100 µL PCR Full Skirt on 1, D1 of NEST 96 Well Plate 100 µL PCR Full Skirt on 1, E1 of NEST 96 Well Plate 100 µL PCR Full Skirt on 1, F1 of NEST 96 Well Plate 100 µL PCR Full Skirt on 1, G1 of NEST 96 Well Plate 100 µL PCR Full Skirt on 1, H1 of NEST 96 Well Plate 100 µL PCR Full Skirt on 1]
```
Similarly, a `TypeError` is also observed when using any the following commands as well:
``` 
p_multi.dispense(100, plate_96well.columns_by_name()['1'] )
p_multi.mix(100, plate_96well.columns_by_name()['1'] )
p_multi.pick_up_tip(plate_96well.columns_by_name()['1'] )
p_multi.return_tip(plate_96well.columns_by_name()['1'] )
```

# Expected behavior

Accessing a column by name when working with the multi-channel pipette should produce the expected robot movement rather than a `TypeError`

nickcrider, 2020-05-01:
The fact that this doesn't work I am guessing is related. Commenting here instead of opening a new ticket.
```python
p50.transfer(50, 
     reaction_plate.columns_by_name()['1'].bottom(), 
     mag_plate.columns_by_name()['1']
)
```

IanLondon, 2020-05-21:
Thread for more context: https://opentrons.slack.com/archives/CL08J5DJ7/p1589900542000700

closed 2022-09-30 by bndo