Opentrons/opentrons #5463: bug: App spuriously disconnects from the OT-2 after connecting to it for the first time after restarting it

SyntaxColoring, 2020-04-22:
# Current behavior

After successfully connecting to an OT-2 that has just restarted, the connection usually only lasts a few seconds before it is cut off. Then, after a few seconds of disconnection, it comes back on its own.

I don't think this is associated with an actual connection loss.

# Steps to reproduce

1. Connect to an OT-2 over USB.
2. Click **Restart.**
3. Wait for the OT-2 to reappear. When it does, connect to it.
4. Wait a few seconds. You can pass the time by flicking the lights on and off, to prove that the connection really works.
5. The OT-2 will disconnect with a "Your robot is advertising its IP address but not responding to requests" message.
6. Wait a few seconds. The OT-2 will become reconnectable on its own.

This may not happen every time, but for me, it happens at least most of the time.

# Expected behavior

After successfully establishing a connection to the OT-2, the connection should not break unless there are underlying connectivity issues.

mcous, 2020-04-22:
Important to note: at this stage in the user flow, there is no persistent connection to the OT-2, so no connection is being dropped here.

From your description, this is probably the result of stale `GET /health` requests from discovery client timing out at the OS default (like 60 seconds to 2 minutes I think?) and causing a spurious health check failures to bubble up before a subsequent health check from the next polling cycle re-asserts the proper health value.

If that race condition actually exists (pretty sure it does) then it's an easy patch

mcous, 2020-08-10:
@SyntaxColoring we should revisit this issue in light of #6193, which put protections in place to throw away the kind of failed health check responses that could have lead to this bug. If the race conditions I described above were, in fact, the cause of these flickers, then 3.20.0 will fix this issue

SyntaxColoring, 2020-08-11:
Groovy.

I unfortunately don't have an OT-2 to test with, so someone else will probably have to check.

SyntaxColoring, 2020-10-14:
Mostly a note to self:

* When I first repro'd and ticketed this issue, I was on v3.16.1.
* #6193, which we think fixed this, landed in v3.20.0 (according to me scanning the Git tags on https://github.com/Opentrons/opentrons/commit/0e089a7347485e45b2c7ca58d46a0b83cf9901ed).

SyntaxColoring, 2020-10-22:
Added "turn off your computer's Wi-Fi" to the beginning of the reproduction steps.

SyntaxColoring, 2020-10-22:
This seems fixed to me.

* Could reproduce with robot v3.21.2 and a debug build of app v3.16.1, after 2-3 attempts.
* Couldn't reproduce with robot v3.21.2 and app v3.21.2, after 5-6 attempts.

mcous, 2020-10-23:
Also took a crack at this and I was unable to reproduce after 3 tries on 3.16.1 nor 3.21.2. Give @SyntaxColoring's promising test results yesterday (with actual repro), gonna close this one out. I think the issue analysis makes sense and that race condition was accounted for the in 3.20 discovery client rewrite

closed 2020-10-23 by mcous