Opentrons/opentrons #5464: feat: Allow for shifted OpenTrons thermocycler module

Koeng101, 2020-04-22:
# Overview

I am looking to do a long / complicated protocol with many samples. In order to do this, I need to use all deck locations on the OT2 deck. Originally, you could shift the Thermocycler to only access 72 wells on the thermocycler in order to access deck location 11 and 8. However, it seems that this is disabled in software:

`ValueError: Could not load NEST 96 Well Plate 100 µL PCR Full Skirt on 11 as deck location 11 is obscured by Thermocycler Module on 7`

Physically, there shouldn't be a reason I can't do this, since the thermocycler would simply be shifted. I would like a way to override this constraint so that I can have access to more of the deck.

# Implementation details
The only change I would like is a way to define the thermocycler in such a way that I can use slot 11 and slot 8. This can be either removing the constraint, or more likely, adding an optional parameter allowing me to access the thermocycler in this way. 

# Design

# Acceptance criteria
- [ ] Ability to simulate runs with a shifted thermocycler
- [ ] Ability to physically run with a shifted thermocycler

SyntaxColoring, 2020-04-23:
Hey @Koeng101! Thanks for reaching out.

The Thermocycler fits into slots 7 and 10, but it has an overhang that physically occludes the leftmost parts of slots 8 and 11.

Are you talking about physically shifting the whole Thermocycler slightly to the left, to remove that occlusion? If the Thermocycler were shifted, how would you lock it in place on the deck?

Koeng101, 2020-04-23:
> Are you talking about physically shifting the whole Thermocycler slightly to the left, to remove that occlusion?

Yes

> If the Thermocycler were shifted, how would you lock it in place on the deck?

I couldn't find a picture, but on the bottom of the thermocycler module there is line in the metal so that you can lock it in either 72 or 96 well positions

SyntaxColoring, 2020-04-23:
> I couldn't find a picture, but on the bottom of the thermocycler module there is line in the metal so that you can lock it in either 72 or 96 well positions

Oh, interesting. I didn't know about that, and I'm cooped up without a Thermocycler at the moment.

I'll ask internally to see why the ability to use slots 8 and 11 was removed, and if it can be added back in.

Koeng101, 2020-04-23:
Awesome thank you

SyntaxColoring, 2020-04-24:
Shifting the Thermocycler to free up slots 8 and 11 was a Thermocycler beta feature that was removed for the final production release. We removed it because we considered it a mere nice-to-have, and we didn't think it would have been worth the complexity to support. But we did leave room to add it back in, when the time comes.

Astute users might notice that newer Thermocyclers have a metal bar on their underside that physically prevents you from shifting them like this. That bar is removable; if this feature were added back, we would publish instructions for removing it.

This is a good feature request, and I can definitely see how freeing up those slots would help on a crowded deck. For now, though, I think it will be lower-priority for us.

-----

Other folks: Hello! If you need this feature, please chime in here or reach out to Opentrons Support, so we can prioritize it accordingly. 🙂

Koeng101, 2020-04-24:
Ok, then how about work arounds? (I need this feature) For example, could I just 
```
        def get_item_covered_slot_keys(sk, i):
            if isinstance(i, ThermocyclerGeometry):
                return(set([7, 8, 10, 11]))
```
Remove 8,11 from this list, then reinstall the python library on raspberry pi on the robot itself?

AFAIK looking through the code, that is the place that hard codes this constraint.

b-cooper, 2020-04-26:
Hi @Koeng101, that workaround is reasonable. This will allow you to load items into slots 8 and 11 on the deck, but is dangerous in that it doesn't direct the the rest of the geometry system to the shift of the thermocycler and its wells. 

If you do choose to work with this in the short term, know that specifying 'A1' on the labware in the thermocycler will actually point the pipette tip in the vicinity of the well 'A4', there may also be a small offset in the x direction that you'll need to compensate for (probably via labware calibration). As such, accessing all wells in the thermocycler's contained labware from a protocol will actually address the well 3 columns to the right of that well.  This also means that addressing any wells in columns '10', '11', or '12', of the thermocycler's contained labware, will result in a possible collision with the thermocycler itself. This workaround will provide no safety assurance for those cases and it is the protocol authors responsibility to avoid them.

Full support for this configuration is something that we're planning to squeeze into an upcoming release and I will update this thread with details as they fall into place.

Koeng101, 2020-04-27:
Got it - I can make sure the geometry works out. 

I also worked out some hacks with sample format + tip reuse so I might not need this feature. It, however, would be really useful for later protocols.

b-cooper, 2020-04-27:
@Koeng101 Sounds good. It looks like we will probably have full support for it by the time "later protocols" come around. 

If you do end up trying to implement the shifted set up of the thermocycler, there is one other thing to be very careful about, that I didn't remember to include above. In the shifted position, when the thermocycler's lid is closed, direct movements between slots 1 or 4 to slots 8 or 11 risk crashing pipette tips into the lid of the thermocycler. 

There is a yet unreleased commit that provides crash avoidance behavior for dangerous moves around the closed lid, https://github.com/Opentrons/opentrons/pull/5390. It will unfortunately not immediately cover the shifted position and would require either appending to the list of dangerous moves added in that PR, or explicitly moving to a safe intermediate point when travelling around the closed thermocycler lid.

PR #5498 (2020-04-28): feat(api): Allow Non-Full plate Thermocycler Configuration
## overview

Closes #5464. This PR adds in support for a "semi" thermocycler configuration which can support up to 80 samples.

## changelog

- Add in more 'bad locations' in geometry to avoid when a thermocycler is loaded onto the deck
- Add in a parameter to the `load_module` method called `configuration` that allows you to specify the non-full plate configuration by setting the value to `semi`
- Modify the thermocycler offset when the semi configuration is specified
- Modify the wells you can access on the plate (this will get a lit [... cut]
closed 2020-04-28 by Laura-Danielle