Opentrons/opentrons #5471: bug: When calibrating a tip rack, retrying tip pickup changes the return height

SyntaxColoring, 2020-04-22:
# Overview

When you're calibrating a tip rack, after the pipette picks up a tip, the app asks you if it succeeded.

If you answer **No, try again,** the tip will be returned higher than normal.

# Current behavior

If I answer **No, try again:**

[image]

If I answer **Yes, save calibration:**

[image]

# Steps to reproduce

1. Upload the test protocol below.
2. Start calibrating a tip rack.
3. Select **No, try again.** Observe the tip return height.
4. Select **Yes, save calibration.** Observe the tip return height.

```python
metadata = {"apiLevel": "2.2"}

def run(protocol):
    tip_racks = [protocol.load_labware("opentrons_96_filtertiprack_20ul", 1)]
    pipette = protocol.load_instrument("p20_single_gen2", mount="right", tip_racks=tip_racks)
    pipette.pick_up_tip()
    pipette.return_tip()
```

# Expected behavior

The tip return height should be the same, whether I answer **Yes, save calibration** or **No, try again.**

I think the height that's used for **Yes, save calibration** is the correct one to use. That appears the same as what `return_tip()` uses in an actual run.

# See also

#5468

nusrat813, 2020-04-24:
The issue does not persist in 3.17.0, API level 2.3+

SyntaxColoring, 2020-04-24:
^ To clarify, it *does* persist in v3.17.0 *APIv2.2*, so we're keeping this issue open.

closed 2022-03-04 by mcous