Opentrons/opentrons #5473: bug: Labware calibration adjustments for all tip racks but the last are ignored

SyntaxColoring, 2020-04-23:
# Overview

During the tip rack calibration flow, if you make an adjustment to one tip rack, that adjustment doesn't carry forward. The next tip rack will start from scratch.

This is a little scary, because making sure that the adjustments carry forward was specifically the fix for a big calibration bug.

# Background

## One offset per labware type

On the OT-2, all the instances of a given labware type share the same calibration offset. Although that might be problematic and confusing, that's a _separate_ issue.

This issue is about how that's implemented with tip racks.

## Issue #4288

There used to be a bug, #4288, that caused problems if you had lots of tip racks.

First: Adjustments made to tip racks earlier in the calibration flow wouldn't propagate to tip racks later in the calibration flow. So, if you had 5 tip racks that were each 1 mm too far right, you'd make that same 1 mm adjustment each time.

Second: The code that collapsed all the tip rack adjustments into one final offset assumed that the adjustments *would* be propagated.

Combined, this meant that the final offset would be the *sum* of all the tip rack adjustments you made. You'd end up with an offset of 5 mm instead of 1 mm, causing the pipette to try to pick tips up from entirely the wrong place.

The fix (#4322) was to make the adjustments propagate from one labware to the next. From the fix's [test notes](https://github.com/Opentrons/opentrons/pull/4322#issuecomment-547976435):

> The differing calibration carried forward, and successive differing calibrations remained independent and propagated, rather than being summed.

# Current behavior

We seem to have regressed on the fix, #4322. Adjustments made to earlier tip racks no longer apply to later tip racks.

But, we appear *not* to have regressed on the original issue, #4288. (I only tested quickly, on APIv2.2 and v3.16.1.)

Weird!

# Steps to reproduce

1. Upload a protocol with lots of tip racks. I used an APIv2.2 protocol on robot software v3.16.1.
2. Make an obvious adjustment to the first tip rack.
3. Move on to the next tip rack. Observe that the default position is unaffected by the adjustment you made to the first tip rack.
4. Repeat for the remaining tip racks.
5. Run the protocol. Observe that no adjustments except the one made to the last tip rack had any effect.

# Expected behavior

~¯\\\_(ツ)\_/¯~

During labware calibration, tip rack adjustments should propagate from one tip rack to the next.

silvtal, 2021-09-27:
I believe we might be having this problem? I've been recalibrating everything for hours and I'm out of ideas at this point.

I first thought it was a custom tiprack problem: we tried to used two different tipracks - the one at module 7 was going to be used by the p300 and m300. The one at module 8 was foinf to be used by the m300 only. During pre-protocol calibration, the p300 was calibrated with the tiprack at module 7 and the m300 used the one at module 8. But then during the protocol run, the m300 failed completely when picking from the tiprack at module 7.

So I re-measured everything from the custom tipracks and used a caliper, but I tried a script like this and the problem persisted:
```

    for pipette in [m300,p300]:
        ctx.comment("Now for pipette "+str(pipette))
        for w in ["A1", "A12"]:
            ctx.comment("Now for well "+w)
            pipette.pick_up_tip(tiprack_200_7[w])
            pipette.return_tip()
            pipette.pick_up_tip(tiprack_200_8[w])
            pipette.return_tip()
            pipette.pick_up_tip(tiprack_200_9[w])
            pipette.return_tip()
```
The pre-protocol calibration calibrates all tips racks with the p300, but then m300 has some offset and tries to pick too far right or left, to the point that the tiprack itself moves from the module. I thought it might be an offset problem (#7878).

 Then I had the idea of trying my test script with standard tips... And it failed as well. Then, it really was that #7878 problem? I tried the fix mentioned at that thread, and it kind of works, but it only works for the standard or for the custom tipracks, not both, obviously. So it doesn't suit my needs. 

I now believe it might be this other issue (#5473), because I changed my script like this:

```
    pipette=m300
    ctx.comment("Now for pipette "+str(pipette))
    for w in ["A1", "A12"]:
        ctx.comment("Now for well "+w)
        pipette.pick_up_tip(tiprack_200_7[w])
        pipette.return_tip()
        pipette.pick_up_tip(tiprack_200_8[w])
        pipette.return_tip()
        pipette.pick_up_tip(tiprack_200_9[w])
        pipette.return_tip()
        
    pipette=p300
    ctx.comment("Now for pipette "+str(pipette))
    for w in ["A1", "A12"]:
        ctx.comment("Now for well "+w)
        pipette.pick_up_tip(tiprack_200_7[w])
        pipette.return_tip()
        pipette.pick_up_tip(tiprack_200_8[w])
        pipette.return_tip()
```
so that the m300 would be calibrated with the tiprack at module 9... And this time the failure was even more catastrophic. The m300 couldn't even pick tips from the A12 well at module 7. All this using standard tips.

Any ideas to fix this?

Matt-Zwimpfer, 2021-10-04:
@silvtal Thanks for bringing this to our attention. For your specific case, it would best to[ contact us at Opentrons Support](https://support.opentrons.com/en/articles/789515-getting-help-from-opentrons-support) so we can investigate this further.

closed 2022-03-04 by mcous