Opentrons/opentrons #5481: bug: labware-creator does have GEN2 pipettes available in test protocol

nickcrider, 2020-04-23:
# Overview

The GUI version of the labware-creator hosted at https://labware.opentrons.com/create cannot export test protocols with GEN2 pipettes.

# Current behavior

Python protocols for testing custom labware exported by the labware creator can only contain GEN1 pipettes which are hardcoded to be only on the right mount. Customers who have purchased only GEN2 pipettes (and perhaps have already mounted it on the left mount) need to ask support to modify the python code produced by the GUI in order to be able to run the test file on the robot without errors.

# Steps to reproduce

Have a robot on 3.16.1 with any GEN2 single mounted on any mount. Create custom labware according to the instructions.  Attempt to run the test python file produced by the tool. The Run App will report the required pipette missing.

# Expected behavior

The test code produced by the GUI should not require modification by hand in order to successfully run.

SyntaxColoring, 2020-04-24:
This is fixed in edge by #5084. Labware Library will need a release before users can benefit from that, though.

I'm not sure if anything is blocking the next Labware Library release, or if it's just not something we've gotten around to.

nickcrider, 2020-04-24:
Ah I did not catch that in my search of existing issues. Will go ahead and close this ticket then. Thanks!

closed 2020-04-24 by nickcrider