Opentrons/opentrons #5504: spike: Investigate TC command mapping in executor

Kadee80, 2020-04-28:
# Overview

This ticket covers the research needed to be done to determine the scope of the executor work for TC.

# Implementation details

Investigate mapping of the following api calls:

  - [ ] openLid
  - [ ] closeLid
  - [ ] activateLid
  - [ ] deactivateLid
  - [ ] activateBlock
  - [ ] deactivateBlock
  - [ ] runProfile

# Acceptance criteria

Clear path forward for beginning executor work.

IanLondon, 2020-04-29:
## How to error out when unsupported asynchonous behaviors are specified in protocol commands

The language of the v4 JSON TC commands is async with "setX" and "awaitX" vocabulary (except for open/close lid which is sync. However, API and TC hardware itself currently only support sync TC temperature commands.

When an await immediately follows an async set, it's equivalent to a sync set. Eg, `thermocycler/setTargetLidTemperature + thermocycler/awaitLidTemperature` = "sync set temp". But if the command immediately after the `set` isn't a corresponding `await` command, the protocol has expressed something the executor is not capable of supporting, so the executor needs to throw an error.

Until we want it to do differently, PD will always do set + immediate await for these TC commands. So this guard is only for user-created JSON (or PD bugs? That'd be a pretty bad bug though to mess up the commands array like that)

One way: `dispatch_json` could first call some `has_async_thermocycler_commands(commands)` helper fn that returns a boolean, and if that boolean is true, raise a `RuntimeError` with a human-readable message (targeted at a user making JSON protocols outside of PD, probably?).

## thermocycler/setTargetBlockTemperature + thermocycler/awaitBlockTemperature

`tc.set_block_temperature(temperature)`. Will block until temp is reached.

_SIDE NOTE: I don't think this applies to us, but you can also set and await. `tc.set_block_temperature(4, hold_time_seconds=15, hold_time_minutes=45, block_max_volume=50)`_

## thermocycler/setTargetLidTemperature + thermocycler/awaitLidTemperature

`tc.set_lid_temperature(temperature)`. Will block until temp is reached.

## thermocycler/runProfile + thermocycler/awaitProfileComplete

```py
profile = [
  {temperature: 10, hold_time_seconds: 30},
  {temperature: 60, hold_time_seconds: 30}]

tc.execute_profile(steps=profile, repetitions=100, block_max_volume=30)
```

**How JSON command params plug into Python fn args:**
`params.volume` can plug into `block_max_volume`
`params.profile` maps to `steps=profile` (we need to change `params.profile[i].holdTime` to `hold_time_seconds`, it is seconds just different key name)
`repetitions` kwarg is not relevant, we just do repetitions in the profile array itself.

## thermocycler/deactivateBlock

`tc.deactivate_block()`

## thermocycler/deactivateLid

`tc.deactivate_lid()`

## thermocycler/openLid

`tc.open_lid()`

## thermocycler/closeLid

`tc.close_lid()`

sfoster1, 2020-04-29:
We should probably do a first pass over the protocol that coalesces commands like that and emits some intermediate command stream, that should handle the problem without having to make `dispatch_json` stateful

b-cooper, 2020-04-29:
This all looks good and I second Seth's comment.  Keeping the `dispatch_json` stateless is a good idea.  We may have to do this level of command-order validation in the future, so it makes sense to me to generalize our approach.

closed 2020-05-05 by IanLondon