Opentrons/opentrons #5537: bug: Redux DevTools fail to install on Windows

dozercodes, 2020-04-30:
# Overview

Redux dev tools fail to install on Windows. As a result I cannot view the state of USB devices. 

# Current behavior

screenshot:
[image]

Log output:
```$ "C:\Users\Storm\AppData\Local\Programs\Opentrons\Opentrons.exe" --devtool --reinstallDevtools
18:54:40.722 [log] info: Level "error" and higher logging to C:\Users\Storm\AppData\Roaming\Opentrons\logs\error.log
18:54:40.728 [log] info: Level "debug" and higher logging to C:\Users\Storm\AppData\Roaming\Opentrons\logs\combined.log
18:54:40.728 [log] info: Level "info" and higher logging to console
18:54:40.951 [main] info: Starting App
18:54:40.968 [ui] info: Loading file://C:\Users\Storm\AppData\Local\Programs\Opentrons\resources\app.asar\ui\index.html
18:54:40.999 [main] info: Creating renderer logger
(node:12368) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 unpipe listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:12368) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 unpipe listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:12368) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:12368) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:12368) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 unpipe listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:12368) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 unpipe listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:12368) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [File]. Use emitter.setMaxListeners() to increase limit
(node:12368) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [File]. Use emitter.setMaxListeners() to increase limit
18:54:41.340 [log] info: Level "error" and higher logging to C:\Users\Storm\AppData\Roaming\Opentrons\logs\error.log
18:54:41.341 [log] info: Level "debug" and higher logging to C:\Users\Storm\AppData\Roaming\Opentrons\logs\combined.log
18:54:41.341 [log] info: Level "info" and higher logging to console
18:54:41.744 [update] info: Checking for update
(node:12368) UnhandledPromiseRejectionWarning: TypeError: Le.a.command is not a function
    at C:\Users\Storm\AppData\Local\Programs\Opentrons\resources\app.asar\lib\main.js:1:38034
    at Ue (C:\Users\Storm\AppData\Local\Programs\Opentrons\resources\app.asar\lib\main.js:1:38305)
    at Array.map (<anonymous>)
    at C:\Users\Storm\AppData\Local\Programs\Opentrons\resources\app.asar\lib\main.js:1:39059
(node:12368) UnhandledPromiseRejectionWarning: TypeError: Le.a.command is not a function
    at C:\Users\Storm\AppData\Local\Programs\Opentrons\resources\app.asar\lib\main.js:1:38034
    at Ue (C:\Users\Storm\AppData\Local\Programs\Opentrons\resources\app.asar\lib\main.js:1:38305)
    at Array.map (<anonymous>)
    at C:\Users\Storm\AppData\Local\Programs\Opentrons\resources\app.asar\lib\main.js:1:39059
(node:12368) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:12368) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:12368) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
(node:12368) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
18:54:41.809 [renderer] info: Rendering app UI
18:54:42.222 [main] warn: Failed to install devtools extension {
  name: 'REACT_DEVELOPER_TOOLS',
  forceReinstall: true,
  undefined: [ { name: 'REACT_DEVELOPER_TOOLS', forceReinstall: true } ]
}
18:54:42.235 [main] warn: Failed to install devtools extension {
  name: 'REDUX_DEVTOOLS',
  forceReinstall: true,
  undefined: [ { name: 'REDUX_DEVTOOLS', forceReinstall: true } ]
}
18:54:43.011 [update] info: Found version 3.17.0 (url: Opentrons-v3.17.0-win-b15169.exe)
18:54:43.016 [renderer] warn: App update available on load, redirecting to app update.
18:54:43.078 [update] info: Checking for update
18:54:43.124 [update] info: Found version 3.17.0 (url: Opentrons-v3.17.0-win-b15169.exe)
```

# Steps to reproduce

1. Install 3.17.1-alpha.1 on a windows machine
2. Start with Dev Tools active
3. Observe

# Expected behavior

Redux DevTools should be installed and "Redux" an available tab in the row of dev tools tabs

mcous, 2020-05-01:
Still in the process of inspecting the build, but all signs point to this being another instance of the `electron-builder` library pulling in the wrong version of a dependency - in this case, `execa`, which we pulled in to read the USB-to-Ethernet driver version via PowerShell

If this is the case, we have a few options ordered by complexity:

1. Drop execa and rely on the built-in Node.js child process library that execa wraps in a (much) nicer interface
2. Somehow isolate the app production build in CI
3. Enable full bundling in the app-shell's webpack config
    - Currently, it's only bundling source while leaving node_modules unbundled
    - Certain dependencies have no responded well to being bundled in the past, which is why the current configuration is what it is

None of those options (except for maybe no. 3) feel particularly long term nor do they feel like they address the root of the problem. I'll be ruminating on longer term stuff after I've inspected the production app bundle on Windows and confirmed the problem

mcous, 2020-05-01:
More full audit of a production build I just did on my mac:

| dependency                  | spec      | actual   | status |
| --------------------------- | --------- | -------- | ------ |
| @thi.ng/paths               | `^1.6.5`  | `1.6.5`  | ok     |
| ajv                         | `^6.10.2` | `6.10.0` | wrong  |
| dateformat                  | `^3.0.3`  | `3.0.3`  | ok     |
| electron-context-menu       | `^0.15.1` | `0.15.1` | ok     |
| electron-debug              | `^3.0.1`  | `3.0.1`  | ok     |
| electron-devtools-installer | `^2.2.4`  | `2.2.4`  | ok     |
| electron-dl                 | `^1.14.0` | `1.14.0` | ok     |
| electron-store              | `^4.0.0`  | `4.0.0`  | ok     |
| electron-updater            | `^4.1.2`  | `4.1.2`  | ok     |
| execa                       | `^4.0.0`  | `0.7.0`  | wrong  |
| form-data                   | `^2.5.0`  | `2.5.0`  | ok     |
| fs-extra                    | `^6.0.1`  | `8.1.0`  | wrong  |
| get-stream                  | `^5.1.0`  | `3.0.0`  | wrong  |
| merge-options               | `^1.0.1`  | `1.0.1`  | ok     |
| node-fetch                  | `^2.6.0`  | `1.7.3`  | wrong  |
| node-stream-zip             | `^1.8.2`  | `1.8.2`  | ok     |
| pump                        | `^3.0.0`  | `3.0.0`  | ok     |
| semver                      | `^5.5.0`  | `5.5.0`  | ok     |
| tempy                       | `^0.3.0`  | `0.3.0`  | ok     |
| usb-detection               | `^4.9.0`  | `4.9.0`  | ok     |
| uuid                        | `^3.2.1`  | `3.3.3`  | ok     |
| winston                     | `^3.1.0`  | `3.2.1`  | ok     |
| yargs-parser                | `^10.0.0` | `10.1.0` | ok     |

The situation is, in a word, bad. I think this pushes me towards option (2) in my above comment

PR #5545 (2020-05-05): fix(app-shell): upgrade Electron, avoid hoisting dependencies, and fix devtools
## overview

The library that we use to build our production Electron app (`electron-builder`) has had a history of selecting incorrect dependencies from the `yarn` workspace for inclusion in the production bundle. This has caused issues in the past (see #3920) and will continue to cause issues if not fixed (see #5537).

A quick audit of the app-shell's spec'd dependencies vs what's actually in the bundle was [fairly alarming](https://github.com/Opentrons/opentrons/issues/5537#issuecomment-622464608).

This PR disa [... cut]
commit f9939254 (2020-05-05): fix(app-shell): upgrade Electron, avoid hoisting dependencies, and fix devtools (#5545)

Closes #5537, closes #5541
closed 2020-05-06 by sfoster1