Opentrons/opentrons #5542: bug:

hainesm6, 2020-05-01:
#Overview

API doesn't tollerate tall custom labware, even those shorter than the 150 mm limit.

# Current behavior

For custom labware, `LabwareHeight > LABWARE_DEF_JSON["dimensions"]["zDimension"]`

This then throws a `LabwareHeightError [line 65]` error when running scripts e.g.

> LabwareHeightError [line 65]: The Cobas V3 24 Well Plate 4300 µL on 3 has a total height of 291.0 mm, which is too tall for your current pipette configurations. The longest pipette on your robot can only be raised to 179.7 mm above the deck. This may be because the labware is incorrectly defined, incorrectly calibrated, or physically too tall. Please check your labware definitions and calibrations.

# Steps to reproduce

```python
import json
from opentrons import protocol_api, types

CALIBRATION_CROSS_COORDS = {
    '1': {
        'x': 12.13,
        'y': 9.0,
        'z': 0.0
    },
    '3': {
        'x': 380.87,
        'y': 9.0,
        'z': 0.0
    },
    '7': {
        'x': 12.13,
        'y': 258.0,
        'z': 0.0
    }
}
CALIBRATION_CROSS_SLOTS = ['1', '3', '7']
TEST_LABWARE_SLOT = '3'

RATE = 0.25  # % of default speeds
SLOWER_RATE = 0.1

PIPETTE_MOUNT = 'left'
PIPETTE_NAME = 'p20_single_gen2'

TIPRACK_SLOT = '5'
TIPRACK_LOADNAME = 'opentrons_96_tiprack_20ul'

LABWARE_DEF_JSON = """{"ordering":[["A1","B1","C1","D1"],["A2","B2","C2","D2"],["A3","B3","C3","D3"],["A4","B4","C4","D4"],["A5","B5","C5","D5"],["A6","B6","C6","D6"]],"brand":{"brand":"Cobas v3","brandId":["LONDON BIOFOUNDRY"]},"metadata":{"displayName":"Cobas V3 24 Well Plate 4300 µL","displayCategory":"wellPlate","displayVolumeUnits":"µL","tags":[]},"dimensions":{"xDimension":127.76,"yDimension":85.48,"zDimension":101},"wells":{"A1":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":18.88,"y":69.74,"z":6},"B1":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":18.88,"y":51.74,"z":6},"C1":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":18.88,"y":33.74,"z":6},"D1":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":18.88,"y":15.74,"z":6},"A2":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":36.88,"y":69.74,"z":6},"B2":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":36.88,"y":51.74,"z":6},"C2":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":36.88,"y":33.74,"z":6},"D2":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":36.88,"y":15.74,"z":6},"A3":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":54.88,"y":69.74,"z":6},"B3":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":54.88,"y":51.74,"z":6},"C3":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":54.88,"y":33.74,"z":6},"D3":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":54.88,"y":15.74,"z":6},"A4":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":72.88,"y":69.74,"z":6},"B4":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":72.88,"y":51.74,"z":6},"C4":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":72.88,"y":33.74,"z":6},"D4":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":72.88,"y":15.74,"z":6},"A5":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":90.88,"y":69.74,"z":6},"B5":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":90.88,"y":51.74,"z":6},"C5":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":90.88,"y":33.74,"z":6},"D5":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":90.88,"y":15.74,"z":6},"A6":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":108.88,"y":69.74,"z":6},"B6":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":108.88,"y":51.74,"z":6},"C6":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":108.88,"y":33.74,"z":6},"D6":{"depth":95,"totalLiquidVolume":4300,"shape":"circular","diameter":15,"x":108.88,"y":15.74,"z":6}},"groups":[{"metadata":{"displayName":"Cobas V3 24 Well Plate 4300 µL","displayCategory":"wellPlate","wellBottomShape":"u"},"brand":{"brand":"Cobas v3","brandId":["LONDON BIOFOUNDRY"]},"wells":["A1","B1","C1","D1","A2","B2","C2","D2","A3","B3","C3","D3","A4","B4","C4","D4","A5","B5","C5","D5","A6","B6","C6","D6"]}],"parameters":{"format":"irregular","quirks":[],"isTiprack":false,"isMagneticModuleCompatible":false,"loadName":"cobasv3_24_wellplate_4300ul"},"namespace":"custom_beta","version":1,"schemaVersion":2,"cornerOffsetFromSlot":{"x":0,"y":0,"z":0}}"""
LABWARE_DEF = json.loads(LABWARE_DEF_JSON)
LABWARE_LABEL = LABWARE_DEF.get('metadata', {}).get(
    'displayName', 'test labware')

metadata = {'apiLevel': '2.0'}

def uniq(l):
    res = []
    for i in l:
        if i not in res:
            res.append(i)
    return res

def run(protocol: protocol_api.ProtocolContext):
    tiprack = protocol.load_labware(TIPRACK_LOADNAME, TIPRACK_SLOT)
    pipette = protocol.load_instrument(
        PIPETTE_NAME, PIPETTE_MOUNT, tip_racks=[tiprack])

    test_labware = protocol.load_labware_from_definition(
        LABWARE_DEF,
        TEST_LABWARE_SLOT,
        LABWARE_LABEL,
    )

    num_cols = len(LABWARE_DEF.get('ordering', [[]]))
    num_rows = len(LABWARE_DEF.get('ordering', [[]])[0])
    well_locs = uniq([
        'A1',
        '{}{}'.format(chr(ord('A') + num_rows - 1), str(num_cols))])

    pipette.pick_up_tip()

    def set_speeds(rate):
        protocol.max_speeds.update({
            'X': (600 * rate),
            'Y': (400 * rate),
            'Z': (125 * rate),
            'A': (125 * rate),
        })

        speed_max = max(protocol.max_speeds.values())

        for instr in protocol.loaded_instruments.values():
            instr.default_speed = speed_max

    set_speeds(RATE)

    for slot in CALIBRATION_CROSS_SLOTS:
        coordinate = CALIBRATION_CROSS_COORDS[slot]
        location = types.Location(point=types.Point(**coordinate),
                                  labware=None)
        pipette.move_to(location)
        protocol.pause(
            f"Confirm {PIPETTE_MOUNT} pipette is at slot {slot} calibration cross")

    pipette.home()
    protocol.pause(f"Place your labware in Slot {TEST_LABWARE_SLOT}")

    for well_loc in well_locs:
        well = test_labware.well(well_loc)
        all_4_edges = [
            [well._from_center_cartesian(x=-1, y=0, z=1), 'left'],
            [well._from_center_cartesian(x=1, y=0, z=1), 'right'],
            [well._from_center_cartesian(x=0, y=-1, z=1), 'front'],
            [well._from_center_cartesian(x=0, y=1, z=1), 'back']
        ]

        set_speeds(RATE)
        pipette.move_to(well.top())
        protocol.pause("Moved to the top of the well")

        for edge_pos, edge_name in all_4_edges:
            set_speeds(SLOWER_RATE)
            edge_location = types.Location(point=edge_pos, labware=None)
            pipette.move_to(edge_location)
            protocol.pause(f'Moved to {edge_name} edge')

        set_speeds(RATE)
        pipette.move_to(well.bottom())
        protocol.pause("Moved to the bottom of the well")

        pipette.blow_out(well)

    set_speeds(1.0)
    pipette.return_tip()

```

# Expected behavior

- LabwareHeight = LABWARE_DEF_JSON["dimensions"]["zDimension"]
- The above script should run without a `LabwareHeightError [line 65]`.

mcous, 2021-12-10:
@hainesm6 this protocol is set to `metadata = {'apiLevel': '2.0'}` which is quite old. I don't necessarily think it'll solve the issue, but I would recommend checking this with `{'apiLevel': '2.11'}` just to be safe.

Also, as the error message says...

> This may be because the labware is... incorrectly calibrated

...so the second thing I would try would be a factory reset of your robot's labware calibration data, since this calibration data which affects the total height used during motion planning

hainesm6, 2021-12-10:
Thanks for the suggestions @mcous. As you say it might be because of the `apiLevel` being old but then I created this issue 18 months ago. Unfortunately I now don't have the ability to check if this solves the error. I've therefore closed the issue.

closed 2021-12-10 by hainesm6