Opentrons/opentrons #5551: bug: App doesn't report internal errors during calibration

SyntaxColoring, 2020-05-03:
# Current behavior

As @mcous commented in #4810, during calibration...

> The app collects the error from the API and does not display it.

Instead, it swallows the error silently. This makes things slightly more confusing for users. See #4810 for a concrete example.

# Expected behavior

If something goes wrong during calibration, that should somehow be made clear to the user.

asogluizzo, 2020-11-06:
Check back in on this after 4.0.0 release

SyntaxColoring, 2020-12-04:
According to @mcous, this is a problem with our RPC layer.

In v4.0.0, that RPC layer is no longer used for *deck calibration* or *pipette offset calibration*, so this should no longer be a problem for those.

It's still a problem for *labware calibration.* But labware calibration is going to be overhauled soon, including replacing its RPC layer.

mattwelch, 2021-03-12:
This ticket is now only relevant to labware calibration (as the other calibration stuff is handled by http versus rpc). This ticket is on the way too, as the labware cal project comes to completion.

Any lack of calibration errors using the new infrastructure should get their own tickets.

mcous, 2022-03-04:
Fixed in v5

closed 2022-03-04 by mcous