Opentrons/opentrons #5564: feat: Disallow running multiple instances of the Opentrons App

SyntaxColoring, 2020-05-04:
# Overview

Running multiple instances of the Opentrons App is unsupported. It may cause bad things to happen as the instances step on each other's toes.

On macOS, the operating system will prevent you from running multiple instances. However, Windows and maybe Linux allow it.

Therefore, we should guard against opening multiple instances of the app.

# Acceptance criteria

* [ ] If you try to open a second instance of the app, you're turned away with a helpful message.
* [ ] The lockout mechanism is robust to app crashes or other failures. (For example, if the app crashes before it can delete a lock file, that shouldn't permanently lock me out.)

## Bonus points

* [ ] Also guard against running multiple *installations* of the app simultaneously. (For example, if two versions are installed, or if there's one system-wide installation and one per-user installation.)

mcous, 2020-05-04:
Implementation details: https://www.electronjs.org/docs/api/app#apprequestsingleinstancelock

SyntaxColoring, 2021-12-09:
This can cause confusing error messages when you try to update an OT-2 from one of the running instances.

Here's a video of @emilywools's computer, showing what happens in `~/Library/Application Support/Opentrons` while a production v4.7.0 release is running concurrently with an `edge` `make -C app dev`. It looks like update files are sporadically being deleted from `__ot_buildroot__`.

https://user-images.githubusercontent.com/3236864/145453304-af9f52bd-808c-426b-b4cc-9d777ed90add.mov

Before this video, @emilywools got the following error in the Opentrons App after trying to downgrade a v5.0.0-beta.2 robot to v4.7.0:

> **An error occurred while updating your robot:**
>
> *Error uploading update to robot: ENOENT: no such file or directory,
> open '/Users/<username>/Library/Application Support/Opentrons/\_\_ot\_buildroot\_\_/4.7.0/ot2-system.zip'*

I wasn't able to reproduce that exact error, but I did get these (from two separate attempts):

> **An error occurred while updating your robot:**
>
> *Error uploading update to robot: request to http://10.10.1.250:31950/server/update/Gd6B2nK2SwSFirlcFnYtAw/file failed,
> reason: write EPIPE*

> **An error occurred while updating your robot:**

> *Error uploading update to robot: request to http://10.10.1.250:31950/server/update/kAkCMlfDRVK4WffbV0WA9g/file failed, reason: socket hang up*

I *think* I did this by:

1. Starting a robot server reinstall through Opentrons App v5.0.0-beta.2, with no other instances of the Opentrons App running.
2. Quickly deleting the folder inside `__ot_buildroot__` while the app was in the "Sending update file to robot..." step.

But it's possible I did it with the steps reversed.

Surprisingly to me, these errors didn't pop up immediately. The app spun on the "Sending update file to robot..." for I *think* at least a minute before showing them.

SyntaxColoring, 2021-12-09:
Addition from @mcous:

> Even if we resolved the “multiple instances” question, this is a multiple installations thing. One should never run different installations of the app concurrently (e.g. one in `/Applications`, one from `make dev`). Version mismatches at the Electron level could make bad things happen

closed 2025-12-15 by sfoster1