Opentrons/opentrons #5565: feat: Run App UI should enforce/remind user of valid TC setpoints

nickcrider, 2020-05-05:
# Overview

The Run App currently does not enforce valid setpoints for the Thermocycler, and behaves in a way confusing to users when a temperature outside of the allowed range is requested.

Some users fail to realize the lid can only accept a setpoint `110 >= t >= 37` and try to set the lid below 37C. Ostensibly, they may try invalid temperature settings for the block as well. The App should prevent this.

# Acceptance criteria

Users should be warned and the app should not allow the protocol run to proceed if an invalid setpoint is selected for the Thermocycler.

pantslakz, 2020-05-05:
Recommendation here is to use field validation for module input UI, same as we do in PD.

SyntaxColoring, 2020-05-05:
See #5547 for a similar issue with Temperature Modules.

mcous, 2022-07-26:
Addressed in v6.0.0

closed 2022-07-26 by mcous