Opentrons/opentrons #5591: bug: API suppresses KeyError for bad keys given to labware accessor methods

nickcrider, 2020-05-06:
# Overview

Users expect Python to protect them from supplying a bad key to a dictionary object. Generally, when given a bad key to a dict, python should throw the built-in `KeyError`. This does not currently happen for some labware accessor methods that the docs say should work like dictionary objects.

# Current behavior
The following commands return `[Null]`
- `labware.columns_by_name()['BADKEY']`
- `labware.rows_by_name()['BADKEY']`

Generally this will be caught when fed directly to a movement command as the API will check to see if it's a location, but can result in weird behavior when users try to do something else they think they should be able to do with a standard python dict (eg. iterate over them)

It seems that `labware.wells_by_name()['BADKEY']` does throw the expected `KeyError`, however.

# Steps to reproduce

```python
def run(protocol: protocol_api.ProtocolContext):

    # tipracks
    tiprack_single = protocol.load_labware('opentrons_96_tiprack_1000ul', '11')

    # labware
    plate = protocol.load_labware('nest_96_wellplate_100ul_pcr_full_skirt', '1')
    trough = protocol.load_labware('nest_12_reservoir_15ml', '5')

    # pipettes
    p_single = protocol.load_instrument('p1000_single', 'right', tip_racks=[tiprack_single])

    for well in plate.columns_by_name()['BADKEY']:
        p_single.pick_up_tip()
        p_single.aspirate(100, trough.well(0))
        p_single.dispense(100, well.top(-10))
        p_single.blow_out(well.top(-10))
        p_single.drop_tip()
```
# Expected behavior

The above protocol should produce a `KeyError` in the API, rather than a successful simulation.

nickcrider, 2020-05-07:
In practice, this bug is encountered when users confuse `rows` and `columns` e.g.
```python
for well in plate.columns_by_name()['A']:
        p_single.dispense(100, well)
```

sfoster1, 2020-05-07:
Yeah that's really gross. We'll fix it.

nickcrider, 2020-05-21:
Closed by https://github.com/Opentrons/opentrons/pull/5593

closed 2020-05-21 by nickcrider