Opentrons/opentrons #5624: Add import order linting for JS files

shlokamin, 2020-05-07:
# Overview

As per https://github.com/Opentrons/opentrons/pull/4680#discussion_r363102698 we have an unofficial import order we tend to follow in JS files. Let's have this be checked in a linter so we don't have to worry about this on our own/in PR reviews.

mcous, 2020-05-22:
The only annoying thing here is that the `import/order` eslint plugin does not currently support `import type ...` as it's own import group. See https://github.com/benmosher/eslint-plugin-import/issues/645. That being said, with TypeScript now supporting [the `import type {...}` syntax](https://www.typescriptlang.org/docs/handbook/release-notes/typescript-3-8.html#type-only-imports-and-exports) hopefully there will be some movement soon.

In the mean time, would we want to add this import ordering plugin before the plugin has type import grouping support?

IanLondon, 2020-07-13:
Notes:

Tried out `eslint-plugin-simple-import-sort` in #6028 and tried out eslint `import/order` in #6032. Like Mike mentioned above, neither support making `type` imports their own group and putting them together.

Additional note from slack convo: Mike pointed out

> The eslint-plugin-import project does have an open ticket to figure out type imports, and the ticket became newly relevant once TypeScript added import type syntax in their latest (or second latest, not keeping track) release. Not knowing much about how eslint plugins etc. work in detail, maybe it would be an interesting JS guild project to see if we could put a fork+PR together?

closed 2022-09-30 by bndo