Opentrons/opentrons #5703: bug: set_rail_lights() not working in v3.17.1

opatel0, 2020-05-19:
# Overview
After updating from 3.16.1 to 3.17.1, `opentrons.drivers.rpi_drivers.gpio.set_rail_lights()` does not turn on the OT-2 rail lights and appears to be broken.

# Current behavior
Uploading a protocol with this function in the app returns the following error:
```
Your protocol could not be opened.
AttributeError [line 7]: module 'opentrons.drivers.rpi_drivers.gpio' has no attribute 'set_rail_lights'
```

# Steps to reproduce
Upload a protocol with the following code to a robot with v3.17.1 robot server installed
```
from opentrons import protocol_api
from opentrons.drivers.rpi_drivers import gpio

metadata = {'apiLevel':'2.0'}

def run(ctx):
    light = gpio.set_rail_lights(True)
    tiprack = ctx.load_labware('opentrons_96_tiprack_300ul', 1)
    plate = ctx.load_labware('corning_96_wellplate_360ul_flat', 2)
    pipette = ctx.load_instrument('p300_single_gen2', mount='right', tip_racks=[tiprack])

    pipette.transfer(300, plate['A1'], plate['A2'])
```

# Expected behavior
The expected behavior is for the code to successfully upload and turn on the rail lights during upload (protocol simulation), if the rail lights are turned off before prompting the upload. Note that this occurs successfully on v3.16.1 robot server version.

theosanderson, 2020-05-24:
(I'm another user, not a dev)

This seems likely to be related to https://github.com/Opentrons/opentrons/commit/475ee25cd1aa3d85accae926c69cac30e920d112 (which is IMO a *huge* change). 

Piggybacking on this I think it is really sad if protocols executed through Jupyter (or raw) Python now have no control over GPIO. The point of Jupyter programming is to have more flexibility and if there is some control over the robot that is now impossible that's a great shame and detracts from the openness of OpenTrons.

theosanderson, 2020-05-24:
Your use case can be fixed like this
```
from opentrons import protocol_api
from opentrons.drivers.rpi_drivers import gpio

metadata = {'apiLevel':'2.0'}

def run(ctx):
    light = ctx._hw_manager.hardware.set_lights(rails=True)
    tiprack = ctx.load_labware('opentrons_96_tiprack_300ul', 1)
    plate = ctx.load_labware('corning_96_wellplate_360ul_flat', 2)
    pipette = ctx.load_instrument('p300_single_gen2', mount='right', tip_racks=[tiprack])

    pipette.transfer(300, plate['A1'], plate['A2'])
```

I still am pretty sad if the dependency graph has changed so that some protocol features now can only be used on protocols launched through the OT server, and really hope this isn't a slippery slope towards deprecating Jupyter entirely. It's a use-case that applies to relatively few users but has the potential to open up huge new applications for the robot which could then be rolled out to large numbers of users.

theosanderson, 2020-05-24:
OK, my concerns are significantly assuaged by realising that if you kill (or never start) the OpenTrons server, then everything works as normal.

mcous, 2020-05-25:
@theosanderson Ojas is an Opentrons employee, so this ticket is us flagging the issue for us to fix 😄 

There's absolutely no plan to deprecate Jupyter. As best I can tell, this breakage happened because, for us, the `opentrons.drivers` module is **internal**, so when refactoring our particularly gnarly previous implementation of GPIO, the fact that users may be relying on that import was not considered.

This gets at a larger issue with API stability and why the `ProtocolContext` object of the Python Protocol API v2 was created. For now, we're looking at how to safely expose a stable, public interface for robot GPIO (word of warning, `ctx._hw_manager.hardware.set_lights` should **not** be considered public nor stable) so that all Python protocol users can have a reliable path that won't shift out from under them

theosanderson, 2020-05-25:
Hahaha, sorry for my confusion (had a look at GH profile, but not well enough!)

Super pleased to hear there is not plan to deprecate Juputer :)

And I know I should have no expectations from anything prefixed with an underscore :) I guess I wouldn't have guessed that the internals of the Pi driver were stable either so might have (naively) gone for `ctx._hw_manager.hardware.set_lights`.