Opentrons/opentrons #5728: bug: opentrons_execute does not exit cleanly

nickcrider, 2020-05-22:
# Overview

Protocols executed directly on the robot with the `opentrons_execute` utility perform the expected robot motions, but throws an error instead of killing the process cleanly.

# Current behavior

When used as [described in the docs](https://docs.opentrons.com/v2/new_advanced_running.html#command-line), otherwise valid protocols executed directly on the robot with `opentrons_execute` produce the following error in the terminal:

```
Unhandled error in exception handler
context: {'message': 'Task was destroyed but it is pending!', 'task': <Task pending coro=<Controller.watch_modules() running at usr/lib/python3.7/site-packages/opentrons/hardware_control/controller.py:185> wait_for=<Future pending cb=[<TaskWakeupMethWrapper object at 0x76194fb0>()]>>}
Traceback (most recent call last):
  File "usr/lib/python3.7/asyncio/base_events.py", line 1632, in call_exception_handler
  File "usr/lib/python3.7/site-packages/opentrons/hardware_control/util.py", line 14, in _handle_loop_exception
KeyError: 'exception'
```
Additionally, by monitoring `journalctl -f` in a separate terminal, it appears as if API and serial system logs are not being created when calling `opentrons_execute protocol.py` with the default logging configuration.

Finally, it appears as if the robot does not successfully complete the homing motion upon finishing protocol execution.

# Steps to reproduce
Execute an otherwise valid protocol (Run App throws no errors) with `opentrons_execute` on the robot

```python
from opentrons import protocol_api
metadata = {"apiLevel": "2.0"}

def run(protocol: protocol_api.ProtocolContext):

    # labware
    plate = protocol.load_labware('corning_96_wellplate_360ul_flat', '2')
    plate2 = protocol.load_labware('biorad_96_wellplate_200ul_pcr', '3')

    tiprack = protocol.load_labware('opentrons_96_tiprack_300ul', '1')

    # pipettes
    left_pipette = protocol.load_instrument(
        'p1000_single', 'right', tip_racks=[tiprack] )

    protocol.comment("Loaded")

    left_pipette.transfer(100, plate["A1"], plate2["A2"])
```

# Expected behavior
Protocols that can be executed through the Run App should have the same behavior as when executed via `opentrons_execute` on the robot's command line.

sfoster1, 2020-05-26:
The message appears to be 1. caused by improper teardown of some of the hardware controller's backing tasks in a way that doesn't really harm much and 2. a bug in the function that prints the error, which turns it into a scarier message.

The home behavior doesn't happen for an unrelated reason - that's implemented in the protoocl session machinery rather than the machinery used for `opentrons.execute`.

nickcrider, 2020-05-26:
Should we still expect to see `journalctl` logs created when running with `opentrons.execute` ?

PR #5739 (2020-05-27): fix(api): proper hardware controller teardown
opentrons_execute was printing an exception when it stopped, because we
weren't properly tearing down the module watcher and were relying on the
interpreter to do so - not great. We now explicitly tear down the
hardware control stack.

When we tear down the hardware controller, we cancel some running tasks.
When you cancel a running task, it pends a CancelledException, but the
task needs to execute on the next loop spin, so we need to spin the
loop. This is the complex annoying logic in
thread_manager.__getattribute__.

Closes #5728

## Testing

Ru [... cut]
commit 7faa8da0 (2020-05-27): fix(api): proper hardware controller teardown (#5739)

opentrons_execute was printing an exception when it stopped, because we
weren't properly tearing down the module watcher and were relying on the
interpreter to do so - not great. We now explicitly tear down the
hardware control stack.

When we tear down the hardware controller, we cancel some running tasks.
When you cancel a running task, it pends a CancelledException, but the
task needs to execute on the next loop spin, so we need to spin the
loop. This is the complex annoying logic in
thread_manager.__getattribute__.

Closes #5728
closed 2020-05-27 by sfoster1