Opentrons/opentrons #5734: Allow distribute command to blowout in source well

gcalia, 2020-05-22:
# Overview

When using the aspirate / dispense commands in python you can access the source well for blow out. This is vital when using valuable reagents. However, this cannot be accessed in the distribute/consolidate commands which are necessary for our protocol. 

# Current behavior

blows out in the trash

# Steps to reproduce

<
# imports
from opentrons import protocol_api

# metadata
metadata = {
    'protocolName': ',
    'author': '',
    'description': 'SnAr Analysis Set up',
    'apiLevel': '2.2'
}

# Protocol run function
def run(protocol: protocol_api.ProtocolContext):
    # Labware
    tiprack_1 = protocol.load_labware('geb_96_tiprack_10ul', 7)
    reservoir = protocol.load_labware('usascientific_12_reservoir_22ml', 8)
    plate = protocol.load_labware('corning_384_wellplate_112ul_flat', 5)

    # pipettes
    p10 = protocol.load_instrument('p10_multi', 'left', tip_racks=[tiprack_1])
    p10.flow_rate.aspirate = 50
    p10.flow_rate.dispense = 50
    p10.flow_rate.blow_out = 1000

    p10.pick_up_tip()

    # dye transfer
    for i in range(24):
        p10.distribute(4, reservoir['A1'], plate.columns()[i], new_tip='never')

    p10.drop_tip()

SyntaxColoring, 2020-07-27:
Thanks for creating this issue and providing some context on why this would be important to you.

I think we're already tracking this in issue #5170, which has a couple of partial workarounds. To keep things tidy, I'm going to close this issue. I've also copied your helpful context of the problem into #5170.

If I've missed something that makes this issue not a duplicate, let me know, and I'll reopen this.

SyntaxColoring, 2020-07-27:
Duplicate of #5170

closed 2020-07-27 by SyntaxColoring