Opentrons/opentrons #5741: feat: Allow renaming/nicknaming robots

theosanderson, 2020-05-26:
We are setting up a pipeline in which we will have two OT2 robots. It is obviously crucial we start a protocol on the correct robot! Having the robots named only a long series of hex-digits doesn't make that very easy. So far I find myself flashing the lights on and off a lot. I'm sure I can get in and change this myself, but it seems a sensible feature, either as a nickname in the app or probably better a robot property. I keep thinking I must have missed how to do this so let me know if I have (other than over SSH).

nickcrider, 2020-05-27:
Hey @theosanderson thanks for the feature request! It would make sense to handle this in the App. But for now, look at:
- `/var/serial`
- `/etc/machine-info`
- `/etc/hostname`

sfoster1, 2020-06-22:
@theosanderson Sorry I didn't see this until now - there's actually an undocumented endpoint `/server/name` that will set the name of the robot. Try `POST /server/name`  with an `application/json` body of `{"name": "new-name"}`

theosanderson, 2020-06-22:
Cool!

tanaes, 2020-07-30:
Any tips on accessing this endpoint? When I try to connect from an external machine on the network via e.g. `requests` I'm getting a refusal:

```{python}
r = requests.head("http://192.168.1.4", timeout=3)
```

```
ConnectionError: HTTPConnectionPool(host='192.168.1.4', port=80): Max retries
 exceeded with url: / (Caused by NewConnectionError('<urllib3.connection.HTTPConnection
 object at 0x715cccb0>: Failed to establish a new connection: [Errno 111] Connection refused'))
```

Thanks!

nickcrider, 2020-07-30:
I think you need to use port 31950 for all HTTP endpoints. Is it working on
`http://{robot_ip}:31950/server/name` ?

tanaes, 2020-08-02:
Awesome, that worked for me! Thanks!

If anyone else comes across this, I posted a little Jupyter Notebook showing how on in our protocols library: https://github.com/tanaes/Moeller_Opentrons_protocol_library/blob/master/Misc/rename_robot.ipynb

SyntaxColoring, 2022-07-18:
Implemented in Opentrons App v6.0, under **Robot Settings > Advanced**. 🎉 

We're tracking ongoing work to make the experience nicer, such as allowing non-alphanumeric characters, in the ZenHub epic https://github.com/Opentrons/opentrons/issues/10693. This probably won't be visible to people outside the Opentrons organization—sorry!

closed 2022-07-18 by SyntaxColoring