Opentrons/opentrons #5757: docs_question(api): homing plunger after drop_tip

theosanderson, 2020-05-27:
https://github.com/Opentrons/opentrons/blob/ebb80d31a297ee70ee2a6eee27c41f6bc3441bba/api/src/opentrons/protocol_api/instrument_context.py#L719 says 

```
        .. note::
            OT1 required homing the plunger after dropping tips, so the prior
            version of `drop_tip` automatically homed the plunger. This is no
            longer needed in OT2. If you need to home the plunger, use
            :py:meth:`home_plunger`.
```

This to me implies that the OT2 does not need its plunger homed after dropping the tip, but this function has a `home_after` parameter that defaults to `True` and I think this results in the plunger being homed? The docs seem to contradict the code, as if it not (mostly) needed why would the parameter default to True? My guess is that this note is old documentation and it's now considered safer to home as part of dropping the tip and that's why it does?

nickcrider, 2020-05-27:
Hey @theosanderson thanks for pointing this out! I think this parameter is set as a convenience to the user, as I think the API will throw a must home error if you try to do anything else that requires the position of the pipette plunger to be accurately known, without specifically homing that axis first. But that's just a guess... probably best to let one of the actual engineers chime in here.

As a member of the support team, I can say for sure that we are actively working to update the docs, so I will make a note here and make sure it accurately reflects the robot's behavior.

Also, you might find some additional context here: https://github.com/Opentrons/opentrons/issues/3738

theosanderson, 2020-05-27:
Thanks for this! The issue here is (according to https://github.com/Opentrons/opentrons/blob/7faa8da0d0d1fec30c9447d6208a6bf717fd3700/api/src/opentrons/hardware_control/api.py#L1274) that the plunger might miss steps during ejection and so think it knows its position when it's wrong because of missed steps, and that's why homing may be a good idea.

Homing the plunger adds quite a bit of time to a protocol so it would be useful to know if setting `home_after = False` is likely to have ill effects.

sfoster1, 2020-07-23:
Yeah that note is not correct. I'm pretty sure I'm the one that wrote the note too, really not sure why I wrote that. Homing the plunger is definitely required on the GEN1 pipettes and effectively required on the GEN2s ("effectively" because while they often don't require homing, sometimes they might, and we haven't done any testing on whether it's required at scale).

theosanderson, 2020-07-23:
Cool, thanks for confirmation - as an aside here is a suggestion for how. As an aside, assuming plunger position does not have to be known accurately to pick up pipettes then one could save some time by going from the current order:

- Drop tip
- Home plunger
- Pick up new tip
- Home Z

to

- Drop tip
- Roughly position plunger so it is no longer in eject state
- Pick up new tip
- Simultaneously home plunger & Z

But I can see that's not as elegant.

PR #6203 (2020-07-27): docs(api): remove incorrect note about plunger home
The instrument context docs for drop_tip incorrectly asserted that the
OT-2 does not need to home its plunger after dropping tips, which has
never been true.

Closes #5757
commit 5aba8bd3 (2020-07-27): docs(api): remove incorrect note about plunger home (#6203)

The instrument context docs for drop_tip incorrectly asserted that the
OT-2 does not need to home its plunger after dropping tips, which has
never been true.

Closes #5757
closed 2020-07-27 by sfoster1