Opentrons/opentrons #5765: feat(robot-server): Session Abstraction

amitlissack, 2020-05-28:
# Overview

Create a foundation for all session types. An interface that all session types must support.

Features of a session:
- accept/reject requests to create commands
- execute commands
- enqueues commands for execution
- start executing a command queue
- pause execution
- stop execution
- resume execution
- clean up

# Implementation details

Abstract base class makes sense, but isn't the only solution.

# Design

# Acceptance criteria

mcous, 2020-05-28:
Something that we may want in this base class, too, is the concept of equipment requirements.

A given command may reference a pipette requirement or a labware requirement by a requirement ID, and at command execution time (or possibly session creation time) we need to decide if that requirement is satisfied by the attached hardware and provide a linkage from the requirement ID used in the command to the actual attached hardware (or labware definition)

amitlissack, 2020-05-28:
> Something that we may want in this base class, too, is the concept of equipment requirements.
>
> A given command may reference a pipette requirement or a labware requirement by a requirement ID, and [...]

I had wondered if equipment requirements would be commands.

mcous, 2020-05-29:
> I had wondered if equipment requirements could be commands.

I think there's a definite chance we want this. In a Python Protocol, there's gonna be calls to `load_labware` etc. that will have to affect the state of the session's equipment requirements. Depending on how we want to model those state changes, commands could be a really natural way to do that.

That being said, it might be nice to have a simple `POST /session/:sid/pipettes` with the requirements in the body. It would also be cool to think about what a requirement truly is for a pipette or a module. Right now we're so locked into specifying names or model strings exactly, but what if we could specify volume requirements and/or channels and let the server decide if attached equipment satisfied those capabilities?

amitlissack, 2020-05-29:
Cool. I took the command queuing out of this ticket into a separate ticket. Let's do that with this feature as well.

commit c1fd7e87 (2020-06-09): feat(robot-server): session manager (#5796)

* skeleton of session and session manager
* session manager with tests
* asyncify session and sessionmanager
* fix errors from rebase
* add command_execution package and start check_session implementation.
* Session moved to session_types and renamed BaseSession.
* calibration check session ported to new session manager. integration test works. unit tests need work.
* correct the calibration check creation and implement clean up
* CheckSession tests passing.
* rename CheckBaseSession to CheckSession
* unit tests pass.
* the most recent session is th [... cut]