Opentrons/opentrons #5774: bug: standard labware definitions cause LabwareHeightError with p20 GEN2 pipettes

nickcrider, 2020-05-29:
# Overview

Loading certain standard labware onto a GEN2 Temp Deck and then trying to access it with a p20 GEN2 single or multi creates a `LabwareHeightError` even in simulation with the default deck calibration and robot settings.

The problem seems to be with the NEST labware definitions and not the generics.

# Current behavior

A `LabwareHeightError` is thrown for the following combinations of pipettes and labware:

Pipettes:
- `p20_single_gen2`
- `p20_multi_gen2` 

Labware:
- `opentrons_24_aluminumblock_nest_2ml_snapcap`
- `opentrons_24_aluminumblock_nest_1.5ml_snapcap`
- `opentrons_24_aluminumblock_nest_1.5ml_screwcap`

Other combinations are certainly possible were not exhaustively tried at this time. Accessing these labware with any other pipette seems to work as expected.

# Steps to reproduce

In a freshly installed venv on 3.18.1 run the following with `opentrons_simulate`

```phython
metadata = {'apiLevel': '2.4', }

def run(protocol):

    # labware
    tips_20ul = protocol.load_labware('opentrons_96_filtertiprack_20ul', '3')
    plate = protocol.load_labware('nest_96_wellplate_200ul_flat', '2')

    # pipettes
    mp20 = protocol.load_instrument('p20_single_gen2', 'right', tip_racks = [tips_20ul])

    # modules and associated labware
    temp_mod = protocol.load_module('Temperature Module Gen2', '6')
    temp_labware = temp_mod.load_labware('opentrons_24_aluminumblock_nest_2ml_snapcap')

    mp20.pick_up_tip()
    mp20.aspirate(5, plate['A1'])
```

# Expected behavior

Standard labware definitions (for products sold by Opentrons) should not produce a `LabwareHeightError` in simulation.

nickcrider, 2020-06-01:
Had the chance to dig into this a bit more today. Seems like the `zDimension` is a bit unexpected for the NEST block definitions vs the generic one. The height of the the Generic block is 42mm (corresponding to the actual height of the block in the engineering drawings) whereas the height of the NEST block is greater, and may have included the tops of the tubes when measured?
[image]

@ahiuchingau I think based on the commit info, you created these... is there something I am missing here?

Strange that this has not popped up until now, perhaps it's a change in the geometry system in recent updates that makes this few mm difference into a big deal?

NEST 2mL screwcap **doesn't work**
https://github.com/Opentrons/opentrons/blob/2e4db665f2ef0625b16b440744fbdcd1cb6adc5e/shared-data/labware/definitions/2/opentrons_24_aluminumblock_nest_2ml_screwcap/1.json#L26
`total height of 129.59 mm` ... `raised to 124.65 mm` = 4.93 mm

NEST 2mL snap cap **doesn't work**
https://github.com/Opentrons/opentrons/blob/2e4db665f2ef0625b16b440744fbdcd1cb6adc5e/shared-data/labware/definitions/2/opentrons_24_aluminumblock_nest_2ml_snapcap/1.json#L26
`total height of 123.69 mm` ... `raised to 124.65 mm` = 0.96 mm **very weird** since total height is less than the max height the robot reports the pipette can be raised, the 0.96mm seems like a rounding issue somewhere in the code.

Generic 2mL screwcap **works**
https://github.com/Opentrons/opentrons/blob/2e4db665f2ef0625b16b440744fbdcd1cb6adc5e/shared-data/labware/definitions/2/opentrons_24_aluminumblock_generic_2ml_screwcap/1.json#L22

The strangest thing to me is that I was expecting to do the math and have the difference in the `zDimension` account for total height reported in the `LabwareHeightError` but it does not... I am left pretty clueless at this point.

nickcrider, 2020-06-04:
@Laura-Danielle Assuming this ticket is definitely related to what we found yesterday. For now I've just told customers to use `opentrons_24_aluminumblock_generic_2ml_screwcap` in place of the other taller NEST blocks for building out their code in simulation, as that appears to work with the default deck calibration.

Thanks so much for your help yesterday!

SyntaxColoring, 2020-06-15:
@nickcrider, small note:

> Seems like the zDimension is a bit unexpected for the NEST block definitions vs the generic one. The height of the the Generic block is 42mm (corresponding to the actual height of the block in the [...]

I think it's expected that this definition includes the height of the tubes. Definitions for labware on aluminum blocks are expected to represent the *combination* of the labware and the aluminum block.

PR #5908 (2020-06-18): fix(api): Simulate protocols with a realistic deck transform
## overview

Closes #5774 and closes #5829. Support noticed that the labware height error would often get thrown during simulation for certain combinations of pipettes + labware. This is because simulation does not actually use a valid deck transform for any operation checks.

If the labware height error is going to be useful outside of the robot, we should simulate using a valid transform.

Making this a draft PR because I wanted to get feedback.

## changelog

- Modify tests to get them passing
- Add in a valid deck transform to si [... cut]
closed 2020-06-18 by Laura-Danielle