Opentrons/opentrons #5781: bug(app): software update from file button does not respond after an error

theosanderson, 2020-05-30:
# Overview

If you manually update from file (Browse button) and something goes wrong (e.g. unsigned file, not enough space, etc.) and an error message appears, then after this pressing "Browse" again and selecting a file yields no response at all.

(Not a high priority issue)

theosanderson, 2020-05-30:
Restarting the app restores functionality, suggesting this is indeed an app issue.

mcous, 2020-06-01:
@theosanderson pretty sure I know why this is happening; two questions to confirm:

- Does the button work if you select a _different_ file(name) from the one that cause the error? 
- Rather than a full app restart, does navigating away from page and then back make the button work again?

PR #5789 (2020-06-02): fix(app): clear <input> value after robot update file select
## overview

In a small (code-wise) oversight, we forgot to clear the value of the `<input type="file">` that drives the "robot update from file" flow. Without that value reset (as is present on the protocol upload input), the upload button can become "stuck" if you attempt to select the same file

Fixes #5781 

## changelog

- fix(app): clear <input> value after robot update file select

## review requests

1. Try to upload an "update" file that you know will cause an error (e.g. a random, non-update zip file)
2. After the update fa [... cut]
commit 62372b03 (2020-06-02): fix(app): clear <input> value after robot update file select (#5789)

Fixes #5781
closed 2020-06-02 by mcous