Opentrons/opentrons #5806: bug(docs): example for Location.move() implies it alters state when it does not

theosanderson, 2020-06-03:
The example given for Location.move() in the main labware documentation https://docs.opentrons.com/v2/new_labware.html is:

```
from opentrons import types

metadata = {'apiLevel': '2.2'}

def run(protocol):
     plate = protocol.load_labware(
        'corning_24_wellplate_3.4ml_flat', slot='1')
     plate['A1'].center().move(
        types.Point(x=1, y=1, z=1)) # 1mm up, to the right, and towards the
                                    # back of the robot
```

This implies that this operation will "move" the robot's idea of where plate A1 is by this offset. 

In fact the [function docs](https://docs.opentrons.com/v2/new_protocol_api.html#opentrons.types.Location.move) say that it does not alter state and so the example above would do nothing. A better example would at least involve assignment e.g.:

```
a_bit_off_center = plate['A1'].center().move(
        types.Point(x=1, y=1, z=1)) # 1mm up, to the right, and towards the
                                    # back of the robot
```

theosanderson, 2020-06-04:
(the text of that page should also say that it creates a new point, rather than moving existing point)

SyntaxColoring, 2020-06-07:
Thanks for reporting this. We'll fix the example, probably by doing either:

```python
pipette.move_to(plate['A1'].center().move(types.Point(x=1, y=1, z=1)))
```

or

```python
pipette.aspirate(100, plate['A1'].center().move(types.Point(x=1, y=1, z=1)))
```

-----

As an aside, my personal opinion is that `move` is an inherently misleading name for this method. `a.move(b)` should instead be one of:

* `a.moved_by(b)`
* `a.plus(b)`
* `a + b`

See #4763.

commit 8c11ecd1 (2020-06-25): Correct .move() example.

Use the result of .move() as an adjusted location to which the pipette can be moved, instead of implying .move() does the actual moving.

Fixes #5806.
PR #6002 (2020-06-25): docs(api): correct .move() example

# Overview

Fixes #5806.

# Changelog

* In a code sample, use the result of .move() as an adjusted location to which the pipette can be moved, instead of implying .move() does the actual moving.

# Review requests

None in particular.

# Risk assessment

Low.
commit 5c948710 (2020-06-25): docs(api): correct .move() example (#6002)

Fixes #5806.
closed 2020-06-30 by sfoster1